ArduPilot/ardupilot #4510: ArduRover: Using more servos in both Auto and Manual mode

kiki0000, 2016-07-20:
#### Issue details
I am building an ardupilot controlled RC boat and I would like to use two additional servos in both Auto and Manual mode. If I use "0" at RC channel function I can use the 2 servos in Auto mode but not in Manual and if I set RC Function to Pass Through I can use it only in manual mode but not in Auto mode. How could I solve this problem to can I use servos with both the transmitter and in Auto mode? Please help me.
Thanks. 

#### Version
Ardurover v 2.51 (maybe the latest)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ x] Rover

#### Hardware type
APM 2.6

#### Logs

gmorph, 2016-07-21:
This forum is for issues with the code.  If your asking for help with configuration please post here:
http://discuss.ardupilot.org/c/ardurover
and we will answer your question.

Thanks, Grant.

closed 2016-07-21 by gmorph

kiki0000, 2016-07-21:
Ok, thanks.