ArduPilot/ardupilot #4531: Feature Request: Unique Autopilot ID

d-skinner, 2016-07-24:
It would be great to have some sort of completely unique ID for each autopilot. I know we use SYSID_THISMAV to identify systems through mavlink (which is fine and works well) but unless specifically set differently by the user on each individual autopilot there will be multiple systems set on ID: 1 (as this is the default). This causes a number of headaches, mostly for the ground control software (GCS) as it cannot communicate correctly with any of the systems that have clashing IDs. Even worse is the fact that there is no way for the GCS to detect this clash of IDs and therefore warn the user. I've also found that the logs folder can quickly become unmanageable if this parameter is not set correctly and you've been using a number of different systems over the course of a day or two.

Can we not use the unique MCU chip ID or a number that is randomly generated on first boot and stored in EEPROM?

Ultimately the best solution is to transmit this unique ID in place of the 8bit SYSID_THISMAV, although this will probably break backwards compatibility. So for now, a simpler solution would be to create a new parameter (eg. UNIQUE_SYS_ID). GCS could then "request" this parameter and if more than one different response is received a warning can be displayed.

#### Version
All

#### Platform
All

#### Airframe type
All

#### Hardware type
All

DonLakeFlyer, 2016-07-25:
GCS are already capable of determing whether it is talking to multiple vehicles using the same ID. It can check the UID of the board which comes back from the AUTOPILOT_VERSION message. This is planned to be implemented in QGC 3.1 for example.

iskess, 2016-07-25:
This is related to my request from 2 years ago: https://github.com/ArduPilot/MissionPlanner/issues/347

magicrub, 2016-07-25:
> This is related to my request from 2 years ago:
> ArduPilot/MissionPlanner#347

There's param BRD_SERIAL_NUM
<http://ardupilot.org/plane/docs/parameters.html#brd-serial-num-user-defined-serial-number>
which is inert to the flight system, only used as a user-configurable
abitrary value.

d-skinner, 2016-07-30:
@DonLakeFlyer Thanks for the heads up! I hadn't seen the AUTOPILOT_VERSION message before. Although it really only solves half the problem. It also seems like the uid is not yet implemented in any of the ArduPilot codebases. It's simply set to zero. Maybe this is different for the PX4 codebase?

@magicrub Thanks, I'd missed this parameter. Although it doesn't solve the problem I originally stated. It's use makes sense from a naming point of view rather than a MAVLink point of view. It's only real use case (as far as I can tell) is to ensure the same "name"/serial number is displayed no matter which GCS is used. (ie. If you're using someone else's computer, the GCS could still display the same name without you having to type it in on both computers).

So none of these really solve the main problem as I see it, which is the user still needs to specifically set different SYS_ID's for each autopilot. Current best case scenario is that the GCS can detect a conflict using the AUTOPILOT_VERSION message and warn the operator. I still think we need to move the SYS_ID to some form of unique id (or at least a statistically unique id) so that the absolute bare minimum of user interaction is required to ensure the GCS always has clear communication to each individual system it's connected to.

d-skinner, 2017-05-30:
@tridge Any thoughts on this? I've built a GCS and I'd love to be able to ensure I'm speaking with individual aircraft as well as tie logs to a specific vehicle based on this unique ID.

proficnc, 2017-05-30:
Each STM32 based board has access to a unique serial number that already exists in the code. That would be handy as Mavlink data, and in the logs..

DonLakeFlyer, 2017-05-30:
I believe that is what is returned in AUTOPILOT_VERSION.uid from the PX4 stack. QGC isn't using it yet, but at some point I'll get around to looking at that to de-dupe vehicles with the same parameter based id.

d-skinner, 2017-05-31:
@proficnc I thought this was the case so thanks for the clarification.

@DonLakeFlyer Thanks Don, I know you mentioned this previously. I didn't realise PX4 was doing this already (so thats good) but I'm looking to get this done in ArduPilot as well.

I need to do a little digging around the code again and see if I can find reference to what Phil was talking about or see if this hasn't already been implemented since I opened this request.

magicrub, 2017-05-31:
There's a Param named BRD_serial I think. It's innert. Allows you to set
any arbitrary number there for tracking aircraft. Has no effect on flight.

d-skinner, 2017-06-01:
@magicrub Thanks Tom, you mentioned this before but it really doesn't solve the issue as its not a static unique id. It can be changed.

d-skinner, 2017-06-01:
So I've had a dig around and it seems that the serial number comes from the PX4Util::get_system_id function. This gets used to send a MAV_SEVERITY_INFO message to the GCS right at the beginning of a connection. My next task is to see where it's most appropriate to use this function to make it more accessible (ie. in a MAVLink message rather than just at the start of a connection).

peterbarker, 2017-12-19:
@d-skinner Any news on this?

KapilKumar0862, 2020-10-15:
Hello, community, I am also working on QGC, and My requirement is to read the pixhawk number(Unique ID) and store it to a variable and use it further. How can I read pixhawk Unique ID (Note: I want to read Pixhawk number as sson as I connect my hardware to QGC)

peterbarker, 2020-10-15:
I've created a PR to take advantage of an extension field to pass on the unique system identifier - stm32 MCU serial numbers on ChibiOS, for example.

DonLakeFlyer, 2020-10-15:
> I've created a PR to take advantage of an extension field to pass on the unique system identifier - stm32 MCU serial numbers on ChibiOS, for example.

I thought that was what .uid was for. I think that is what PX4 Firmware is doing but not sure. What is this uid2 thing? Anyone know, looks kind of silly.

peterbarker, 2020-10-15:
On Thu, 15 Oct 2020, Don Gagne wrote:

> I've created a PR to take advantage of an extension field to pass on the unique system identifier - stm32 MCU serial numbers on ChibiOS, for
>       example.
>
> I thought that was what .uid was for. I [...]

Well, the comments are relatively clear, I think.  If .uid2 isn't zero 
(by which I reckon they meant, "the array contains no zero bytes"), use it 
in preference to .uid

The *only* thing it gives you is more bytes for the identifier... which is 
handy, because until it existed there wasn't room for all of the stm32 ID 
stuff IIRC - which is why we didn't fill .uid in previously.

PX4 Firmware *do* appear to be using .uid2

DonLakeFlyer, 2020-10-16:
> The *only* thing it gives you is more bytes for the identifier... which is handy, because until it existed there wasn't room for all of the stm32 ID stuff IIRC - which is why we didn't fill .uid in previously.

Duh, right. Sorry.

PR #15580 (2020-10-21): GCS_MAVLink: fill AUTOPILOT_VERSION.uid2 from get_system_id_unformatted
Closes #4531
closed 2020-10-21 by tridge