ArduPilot/ardupilot #4532: error: '__builtin_bswap16' was not declared in this scope

olliw42, 2016-07-24:
#### Issue details
Just tried to compile ArduCopter master of today and got the error

In file included from c:/Users/Olli/Documents/GitHubZips/betacopterM160724/libraries/AP_HAL/utility/sparse-endian.h:23:0, from c:/Users/Olli/Documents/GitHubZips/betacopterM160724/libraries/AP_RangeFinder/AP_RangeFinder_LightWareI2C.cpp:21:
make[3]: Leaving directory `/c/Users/Olli/Documents/GitHubZips/betacopterM160724/modules/PX4Firmware/Build/px4fmu-v2_APM.build/c/Users/Olli/Documents/GitHubZips/betacopterM160724'
c:/Users/Olli/Documents/GitHubZips/betacopterM160724/libraries/AP_Common/missing/byteswap.h: In function 'uint16_t __bswap_16(uint16_t)':
c:/Users/Olli/Documents/GitHubZips/betacopterM160724/libraries/AP_Common/missing/byteswap.h:11:42: error: '__builtin_bswap16' was not declared in this scope
compilation terminated due to -Wfatal-errors.
make[3]: *** [libraries/AP_RangeFinder/AP_RangeFinder_LightWareI2C.cpp.o] Error 1

the project was clean before, so first make px4-v2 took long, repeating make px4-v2 yielded same error

#### Version
ArduCopter master of today
Windows 7, px4 tool chain version 14, px4 Eclipse, target make px4-v2

regards

OXINARF, 2016-07-24:
We are aware of this and are in the process of updating the Windows installer (to have a newer, required, compiler). You can try the one from here (it is new, from today, so there might be bugs in it): https://www.dropbox.com/s/h30lbyz6imy1t9t/pixhawk_toolchain_installer_v15.exe

olliw42, 2016-07-24:
thx for the quick answer, so, I guess I just failed to see an earlier report, sorry

I guess I just will wait until the v15 becomes official

maybe I may suggest to have a version output along with the "Welcome to the PX4 Toolchain." message in this dos box thing, should be easy to do but would make it easier to identify which version one has currently installed 

THX

rmackay9, 2016-07-27:
Here's the fixed windows installer.  http://firmware.ardupilot.org/Tools/PX4-tools/pixhawk_toolchain_installer_v15.exe

I'll update the wiki shortly to make it point at this new installer.  We need to get the word out to people as well so I will post on facebook and maybe diydrones.

magicrub, 2016-07-27:
What are the changes? Updated g++ version, anything else?

rmackay9, 2016-07-27:
@magicrub, here are the changes:
1. updated the license file to provide links to (at least some of) the underlying component's licenses
2. changed install directory to Pixhawk_Toolchain
3. updated gcc to 4.9.2 (I think).

rmackay9, 2016-07-27:
I've also updated the wiki (I think it's being rebuilt at the moment) so I think these pages will show the correct information shortly.
http://ardupilot.org/dev/docs/building-px4-with-make.html
http://ardupilot.org/dev/docs/editing-the-code-with-eclipse.html#editing-the-code-with-eclipse

closed 2016-07-27 by rmackay9

zhoupian2010, 2016-09-23:
i have meet the same question ,but now it have been solved by updating the toolchain version.