ArduPilot/ardupilot #4533: How does one interact with SITL correctly in terms of data sent to it?

mkschreder, 2016-07-24:
#### Issue details
I'm writing a physics simulator that simulates my quad using a dynamics engine (bullet physics). The idea is to feed ArduPilot SITL with accelerometer and gyro data from the physics simulation. So far I think I'm messing up the sensor data, but don't quite understand what I need to do to fix it. 

I have added my own simple sitl interface that reads packets that arrive. Then I'm trying to fill in accel_body and gyro variables using data that arrives from my physics simulation. One of the problems is that my simulation frame uses the y-up, z-back, x-right frame and I think that I need to feed the simulator with accelerometer data using the x-forward, y-right, z-down coordinate system. 

For now I just want to get at least some sort of flight working so following is an approximation. 
I calculate accelerometer data by rotating vec3(0, 0, -9.82) into the orientation of the simulation. This does not seem right because I think I need to multiply it by inverse of the quad rotation so that it always points towards the ground, but no matter what I still get weird results. For example, I would intuitively want to set the gravity to positive z, but in gcs that shows up as though the quad is upside down. So then accelerometer vector is actually supposed to be pointing up? 

I think I'm totally confused about how to do it correctly so I would appreciate any info that can clear this up for me. 

To summarize the question: what do I need to do to my rotation of the rigid body and it's angular velocity to get accelerometer and gyro vectors that make sense for arducopter, considering that I also need to transform the data into quad coordinate system?

tridge, 2016-07-26:
ArduPilot SITL uses NED, so you do indeed need to use x-forward, y-right, z-down.
One thing that I think you may find useful when developing your own SITL backend is to uncomment this debug code:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_Aircraft.cpp#L154
that will log all the raw accel, velocity and position data. The you can use MAVExplorer on the log to plot the derivative of position against velocity and the derivitive of velocity against acceleration. They should match. 
You can extend the logging for gyros and attitude too.
When writing a SITL backend there are several approaches depending on how many sensors your physics engine provides. The simplest backend would provide just:

- rotational acceleration (in radians/second/second)
- accel_body

and it would call 
```
update_dynamics(rot_accel);
update_position();
update_mag_field_bf();
```
to fill in the rest. Doing that will mean that position, velocity and other key variables will be filled in for you.

mkschreder, 2016-07-27:
Thank you Andrew for all the suggestions. I'm closer to getting it to work now. I can takeoff and I was able to calibrate all sensors in qgroundcontrol, but I'm still not 100% certain if the sensor data that I'm sending to the quad is both sufficient and correct. 

I decided to focus on just accelerometer vector and the angular velocity (gyro). In the ArduPilot code I do this: 

    if(sock.recv(&pkt, sizeof(pkt), 0) == sizeof(pkt)){
        accel_body = Vector3f(pkt.accel[0], pkt.accel[1], pkt.accel[2]); 
        gyro = Vector3f(pkt.gyro[0], pkt.gyro[1], pkt.gyro[2]); 
        rcin_chan_count = 8;            
        for(unsigned c = 0; c < 8; c++) rcin[c] = pkt.rcin[c];  
        send_servos(input);             
    }   

    update_position(); 
    update_mag_field_bf();              
      
    adjust_frame_time(2000);            
    sync_frame_time();                  
    drain_control_socket(); 

I find that the attitude dial in qgroundcontrol lags behind a lot. I do get sensor data almost instantly, but the attitude lags and is very slow to adjust. Can this be because of the data being faulty? 

In my very simple simulation, I produce the accelerometer vector like this: 

glm::quat r1(cos(glm::radians(90.0 / 2)), sin(glm::radians(90.0 / 2)), 0, 0); // rotate 90 deg around x (pitch)
glm::quat r2(cos(glm::radians(90.0 / 2)), 0, sin(glm::radians(90.0 / 2)), 0); // rotate 90 deg around y (yaw)
glm::quat r2ef = r1 * r2;           
    
glm::vec3 accel = r2ef * activeQuad->getAccel(); // getAccel is an accelerometer vector in x-right, y-top, z-back system. 

And gyro is simply the angular velocity, but with swapped axis (I'm experimenting currently with negated values because they seem to have been giving better results): 

    glm::vec3 gyro = activeQuad->getGyro();  
    
    state.gyro[0] = -gyro.z; state.gyro[1] = -gyro.x; state.gyro[2] = -gyro.y; 
    state.accel[0] = accel.x; state.accel[1] = accel.y; state.accel[2] = accel.z; 

My accelerometer vector is currently pointing in the direction straight down from the rigid body but in the body frame, so that if I do bodyRot * acc I get a vector that points straight down in world coordinates and has length of 9.82. My gyro is in rad/s. 

If you look at the screenshot below you can see the acc pointing down (that is after multiplying the actual acc vector with body rotation). But I still have very sluggish controls. I suspect it is either because sim runs too slowly or because values sent are incorrect. I'm currently not modeling exponential thrust so my very simple model outputs thrust in newtons based on servo pwm and a torque force to the sides of the same magnitude for simulating yaw. 

Does anyone have any further suggestions? 

https://s31.postimg.org/gd9fie61n/screenshot.jpg

mkschreder, 2016-07-31:
I got it working but only marginally. I still don't know how to send correct accelerometer and magnetometer values to sitl. But I have got it working ok with the built in sim where I just send input to the copter. 

https://s32.postimg.org/v9gqlgk0l/31_Jul_2016_1.jpg

The simple sim code using bullet and irrlicht is here: https://github.com/mkschreder/quadsim.git

mkschreder, 2016-07-31:
I'm down to the conclusion that ardupilot needs both valid mag and accel data to give proper result. I tried disabling mag updates in internal sim mode and I'm getting similar results in terms of instability as I do when I run from physics simulation. 

How can I verify that it is getting correct mag_bf values?

OXINARF, 2016-08-03:
@mkschreder I see that you have opened a thread in the forum, which is the correct place for this type of discussion, so I'm closing this.

closed 2016-08-03 by OXINARF

mkschreder, 2016-08-03:
Yes. And I have solved this so now everything works. 

@tridge calling update_dynamics would be wrong actually because that runs local simulation and I have an external simulation from which I feed sensor values into the sitl backend. I have also modified update_position so that it does not do timing update and moved that code into a separate function (update_time()) which I found was necessary to execute in order to keep everything running.