ArduPilot/ardupilot #4564: parameter documentation out of date

ctech4285, 2016-07-28:
#### Issue details
The parameter documentation generated dynamically from source for the wiki is out of date it seems

ctech4285, 2016-07-28:
Hmm I see,
page: http://ardupilot.org/copter/docs/parameters.html#brd-can-enable-enable-use-of-uavcan-devices

parameter brd-can-enable should include these eventually(?) 
 *  0 - UAVCAN disabled.
 *  1 - Enabled support for UAVCAN actuators and sensors.
 *  2 - Enabled support for dynamic node ID allocation and firmware update.
 *  3 - Sets the motor control outputs to UAVCAN and enables support for dynamic node ID allocation and firmware update.

OXINARF, 2016-07-28:
@ctech4285 That's not in our code documentation so it wouldn't be generated. There is already a PR opened for that (#4554). Also, any wiki issue should be opened in the proper repo: https://github.com/ArduPilot/ardupilot_wiki

closed 2016-07-28 by OXINARF

proficnc, 2016-07-28:
As mentioned, there is a PR for this, but it's only a couple of days old...