ArduPilot/ardupilot #4577: px4toolchain error on Win, #2

olliw42, 2016-07-30:
**Issue details**

using latest px4toolchain on latest arducopter master produces "no such file or directory" error for uavcan: see here www.olliw.eu/drop/px4toolchainlatest.png

procedure:
deleted ardupilot from my local github, to get fresh master
cloned ardupilot master
uninstalled previous v1.4 px4 toolchain
downloaded and installed latest px4 toolchain from wiki link
set up eclipse as in wiki

to double check I did a couple of times px4-v2 and px4-clean, and also synced my local git repository few times, error remains

**Version**

ArduCopter master of today, freshly cloned
Windows 7, px4 tool chain version of today, px4 Eclipse, target make px4-v2

is that me???

regards

OXINARF, 2016-07-30:
@rmackay9 are you having any problem?

olliw42, 2016-07-30:
so, tried this: 
compiled my "old" code, on which I worked the last days and which is based on AC3.4-rc1, using the new px4 toolchain ... it compiles successfully, and flashes successfully, but I can't then connect anymore with MP !!! When I flash what I got with the v1.4 px4 toolchain everything works just fine. I've repeated that several times, also recompiled a few times, but each time the same: The .px4 produced by the new toolchain doesn't allow me to connect to MP while that produced by the older toolchain works flawlessly. I note, it's for exactly the same source, except that I've incremented a number in the firmware string, i.e. the sources differ by exactly ONE char.

I guess I waste my time here.

Cheers :)

PS: if that should be of any use here the two .px4 files, www.olliw.eu/drop/testpx4s.zip
PPS: MissionPlanner is 1.3.39 build 1.1.6038.12291

olliw42, 2016-07-30:
haven't seen while writing, thx OXINARF for looking into that :)

magicrub, 2016-07-30:
You keep saying new px4 tool chain but you mean the new pixhawk tool chain
v15, right? It's in a whole new directory called /pixhawk_toolchain/
I just installed it on Friday and it works great on master for me.

magicrub, 2016-07-31:
> well, as the dump user I am how should I possibly know what version it is
> and whether I should call it px4 toolchain or pixhawk toolchain ... I
> downloaded using the wiki link and ran an installer
> "pixhawk_toolchain_installer_latest.exe", which installed something in a
> folder "C:\pixhawk_toolchain" as well as created a Windows-Start entry menu
> "PX4 Toolchain", of which I started what is named "PX4 Eclipse" which calls
> "C:\pixhawk_toolchain\toolchain\msys\1.0\px4_eclipse.bat", which starts a
> DOS box telling "Welcome to the PX4 Toolchain" along with an Eclipse
> window, which is what I used.
>
> oh, I forgot, also the time of the download needs to be specified, it is
> of yesterday 17:48MEZ+1
>
> (btw: I would have considered it easier to give it a version number and to
> make that version easily accessible to the user ... but that's not my call,
> right)
>
> glad to hear that it works great for you. I admit I'm envious
>
> :)

The latest is v15. I'm in favor of removing the "latest" filename label in
favour of the actual version numbers for this exact reason.

olliw42, 2016-08-03:
in cooperation with OXINARF the issue got solved, many THX :)

The issue is that in contrast to the previous build systems (up to ac3.4rc1 and/or toolchain v14) for AC3.4-dev and/or the latest toolchain the folder apparently must be a valid git repository. I didn't pay attention to that before, since it just always worked without.

So in my folder I did now:
* delete .git and .github
* git init
* git add *
* git commit -m "whateveryouwant"

I had then initially a slight problem in that I got a compiler error which mentioned gcc....4.7.4. So I went to the respective folder and found three compiler, 4.6.2, 4.7.4, and 4.9.3. I renamed the two older ones to make them unusable - et voila: compiling works, flashing works, connection to MP works (a bit rough though), and the code seems to work. :)

Not sure why I had issues with the older compilers. Not sure why they are included. Anyway.

There had been a bit of discussion if I had all submodules. I had, and this was because I did NOT use the Clone to Desktop as described in the wiki, but used GitHub and cloned using that GUI. GitHub does the submodules thing correctly by itself. 

I thus could suggest to maybe change the docu such as to recommend that procedure, instead of the Clone to Desktop + git submodule update --init --recursive thingy, is a bit easier and more Windows-ish. GitHub is available anyhow since it's the main part of GitHub for Windows.

Thx again to all and for all. :) Cheers, Olli

magicrub, 2016-08-03:
And if on Windows you can use a tool such at source tree and its all done
for you!

olliw42, 2016-08-03:
I'm deeply sorry for having crossed your ways with may ways.
I'll not do again.
Cheers, Olli

closed 2016-08-03 by olliw42