ArduPilot/ardupilot #4578: Extra CAM message on first TRIG

AndKe, 2016-07-30:
#### Issue details
In every flight ,at start I see one more CAM, than expected.

Screenshot shows first TRIG (line 210835), then 21ms after that, a CAM log (too fast to be the actual camera....  THEN, next CAM comes at 220ms after the TRIG - and that's the real one)

later, you can see the typical TRIG to CAM delay.
It always happens to the first picture - and picture only.

[image]

#### Version
AP 3.5.3 

#### Hardware type
3DR Pixhawk

WickedShell, 2016-07-30:
I've seen this as well. It's worth noting that if you do your first TRIG before you've connected the CAM you won't see this problem. It relates to the initial pin manipulation.

AndKe, 2016-07-30:
yes, I do handle it in geotagging script, but it would be nice to fix it right in the software.

WickedShell, 2016-07-30:
Agreed, it actually messes up the Septentrio GPS boards as well.

I suspect it relates to the initial manipulation of the pin into an input. It's also worth noting that I stuck a LED on the pin (for the non high resolution feedback) and found the pin to be flickering between a driving output and an input. This needs to also be addressed.

onurboy, 2016-08-03:
I'm trying to use Septentrio GPS board on my fixed wing and also I'm having problem with CAM trigger log, If you can solve this issue I would appriciate. Thanks.

jmachuca77, 2016-08-23:
@WickedShell How does this affect the Septentrio boards? isn't this just a logging issue?

WickedShell, 2016-08-23:
Iirc it causes the septentrio board to see a extra message as well which
messes up post processing.

I just trig before connecting the feedback

mboland, 2016-08-23:
@WickedShell  Not quite sure what you saying here.
"I just trig before connecting the feedback"

Is it the act of inserting the trig cable into the camera that causes the extra message?
Do you just trigger the camera before connecting (latent charge on pin causing trigger message)

Sorry, your response was a bit cryptic for me to understand.
I have just completed installation of the Septentrio board but have not yet used it.

jmachuca77, 2016-08-23:
@WickedShell Ok, what messages is the Septentrio board using? 

DIGICAM_CONTROL
DO_DIGICAM_CONTROL or 
CAMERA_FEEDBACK

I need to figure out what is being sent out because the original post was about messages on the Dataflash which are not sent out. The messages sent out are different, so if this is happening on both parts finding out the message would help pin point it.

mboland, 2016-08-23:
Oh, I think I see, you are referring to the data flash logs on the FC not the log file on the Septentrio.

WickedShell, 2016-08-23:
The problem I see is that when the Pixhawk first tries to trigger the camera and configures the feedback pin I get 2 feedback events. This really messes up both the pixhawk tagging and septentrio board as they both get a double event with only a single photo being taken. The easiest way I work around this until I get to look at the software is to try and trigger the camera before the camera is connected/turned on. This way I don't get a feedback signal to either board as there is no photo/feedback event.

@jmachuca77 septentrio is split into the feedback signal (looking for a pull down IIRC, I have to check the configurations at this point)