ArduPilot/ardupilot #4580: WAF: compilation database empty

khancyr, 2016-07-31:
@guludo 
Hi,

It seems that compilation database is always empty . 
I try on master, on ubuntu 16.04, gcc 5.4.0 , with or without ninja-build (1.5.1)
I try on another project using gcc and cmake, and it works well, so it seems to be something in ardupilot build.

guludo, 2016-07-31:
Hi!

Could you please share more details? Like the command issued and the output.

Are you really using Waf from our submodule?

khancyr, 2016-07-31:
Yes I am using ardupilot master branch.
from clean workspace ( git clean & reset)
I do : 
waf configure 
````
╭─khancyr@khancyr-N551JQ  ~/Workspace/ardupilot  ‹master› 
╰─$ waf configure
Setting top to                           : /home/khancyr/Workspace/ardupilot 
Setting out to                           : /home/khancyr/Workspace/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for need to link with librt     : not necessary 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 2.7.12 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
'configure' finished successfully (3.691s)
````

Then 
waf rover (for exemple it is same for plane, and copter)
````
[280/283] Compiling APMrover2/system.cpp
[281/283] Compiling APMrover2/test.cpp
[282/283] Linking build/sitl/lib/libAPMrover2_libs.a
[283/283] Linking build/sitl/bin/ardurover
Waf: Leaving directory `/home/khancyr/Workspace/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/khancyr/Workspace/ardupilot/build/sitl
Target         Text     Data  BSS    Total  
--------------------------------------------
bin/ardurover  1010435  1864  28608  1040907

Build commands will be stored in build/sitl/compile_commands.json
'rover' finished successfully (40.471s)
````
no error show . Got message that compilation database is store but ...
````
 cat build/sitl/compile_commands.json 
[]% 
````
Got nothing in  build/sitl/compile_commands.json

OXINARF, 2016-07-31:
Yep, it happens to me too but I haven't checked why.

khancyr, 2016-07-31:
try to remove submodules and re init them but it changes nothing... in fact it build quicker that previously...

khancyr, 2016-07-31:
I remember that it was working previously

khancyr, 2016-07-31:
I use python 2.7.12 . I don't know if it is relate but in config.log there is and error in python check 
````
['/usr/bin/python', '-c', 'import imp;print(imp.get_tag())']
err: Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: 'module' object has no attribute 'get_tag'
````
but it is python related : https://github.com/waf-project/waf/blob/850e9ad5fcdd3d0216298c4d2a990df27b196ef0/waflib/Tools/python.py#L595

khancyr, 2016-07-31:
I found the problem ... it caused by waf here :  https://github.com/ArduPilot/waf/blob/master/waflib/extras/clang_compilation_database.py#L51
but for cxx last_cmd isn"t present since it was deleted in https://github.com/ArduPilot/waf/blob/master/waflib/Task.py#L689
so we get nothing in compilation database. 
I try with lastest waf (so work is do on last_cmd) and it isn't working too. But on Waf master , setting keep_last_cmd at True make compilation_database works again.

closed 2016-07-31 by khancyr