ArduPilot/ardupilot #4617: waf: not working on OSX with clang

gmorph, 2016-08-05:
#### Issue details
I was trying to help Bill Bonney get waf working on OSX.  I run the command
./modules/waf/waf-light build --target bin/arducopter-quad
and get the following issue.

[ 15/353] Compiling libraries/AP_AccelCal/AP_AccelCal.cpp
clang: fatal error: optimization flag '-fno-strength-reduce' is not supported

[ 16/353] Compiling libraries/AP_AccelCal/AccelCalibrator.cpp
Waf: Leaving directory `/Users/grant/uav/ardupilot/build/px4-v2'
Build failed
 -> task in 'ArduCopter_libs' failed (exit status 1): 
	{task 30647840: cxx AP_AccelCal.cpp -> AP_AccelCal.cpp.4.o}
['/usr/bin/clang++', '-DCONFIG_ARCH_BOARD_PX4FMU_V2', '-D__STDC_FORMAT_MACROS', '-D__PX4_NUTTX', '-D__DF_NUTTX', '-g', '-fno-exceptions', '-fno-rtti', '-std=gnu++0x', '-fno-threadsafe-statics', '-DCONFIG_WCHAR_BUILTIN', '-D__CUSTOM_FILE_IO__', '-Wall', '-Werror', '-Wextra', '-Wno-sign-compare', '-Wfloat-equal', '-Wpointer-arith', '-Wmissing-declarations', '-Wno-unused-parameter', '-Werror=format-security', '-Werror=array-bounds', '-Wfatal-errors', '-Werror=unused-variable', '-Werror=reorder', '-Werror=uninitialized', '-Werror=init-self', '-Wframe-larger-than=1024', '-Werror=unused-but-set-variable', '-Wformat=1', '-Wdouble-promotion', '-Werror=double-promotion', '-Wno-missing-field-initializers', '-Os', '-fno-strict-aliasing', '-fomit-frame-pointer', '-funsafe-math-optimizations', '-ffunction-sections', '-fdata-sections', '-fno-strength-reduce', '-fno-builtin-printf', '-fvisibility=hidden', '-include', 'visibility.h', '-mcpu=cortex-m4', '-mthumb', '-march=armv7e-m', '-mfpu=fpv4-sp-d16', '-mfloat-abi=hard', '-nodefaultlibs', '-nostdlib', '-std=gnu++11', '-fdata-sections', '-ffunction-sections', '-fno-exceptions', '-fsigned-char', '-Wall', '-Wextra', '-Wformat', '-Wshadow', '-Wpointer-arith', '-Wcast-align', '-Wundef', '-Wno-unused-parameter', '-Wno-missing-field-initializers', '-Wno-reorder', '-Wno-redundant-decls', '-Wno-unknown-pragmas', '-Werror=format-security', '-Werror=array-bounds', '-Werror=uninitialized', '-Werror=init-self', '-Wfatal-errors', '-fcolor-diagnostics', '-Wno-gnu-designator', '-Wno-inconsistent-missing-override', '-Wno-mismatched-tags', '-Wno-gnu-variable-sized-type-not-at-end', '-Wno-c++11-narrowing', '-Wlogical-op', '-Wframe-larger-than=1300', '-fsingle-precision-constant', '-Wno-error=double-promotion', '-Wno-error=missing-declarations', '-Wno-error=float-equal', '-Wno-error=undef', '-Wno-error=cpp', '--target=arm-none-eabi', '--gcc-toolchain=/usr/local', '--sysroot=/usr/local/Cellar/gcc-arm-none-eabi/20140805/bin/../arm-none-eabi', '-B/usr/local/bin', '-include', 'ap_config.h', '-Ilibraries', '-Ilibraries/GCS_MAVLink', '-I.', '-I../../libraries', '-I../../libraries/AP_Common/missing', '-I../../modules/PX4Firmware/src', '-Imodules/PX4Firmware', '-Imodules/PX4Firmware/src', '-I../../modules/PX4Firmware/src/modules', '-I../../modules/PX4Firmware/src/include', '-I../../modules/PX4Firmware/src/lib', '-I../../modules/PX4Firmware/src/platforms', '-I../../modules/PX4Firmware/src/drivers/boards/px4fmu-v2', '-Imodules/PX4Firmware/src/modules/px4_messages', '-Imodules/PX4Firmware/src/modules', '-I../../modules/PX4Firmware/mavlink/include/mavlink', '-I../../modules/PX4Firmware/src/lib/DriverFramework/framework/include', '-IArduCopter/src/lib/matrix', '-I../../ArduCopter/src/lib/matrix', '-Imodules/PX4Firmware/px4fmu-v2/NuttX/nuttx-export/include', '-Imodules/PX4Firmware/px4fmu-v2/NuttX/nuttx-export/include/cxx', '-Imodules/PX4Firmware/px4fmu-v2/NuttX/nuttx-export/arch/chip', '-Imodules/PX4Firmware/px4fmu-v2/NuttX/nuttx-export/arch/common', '-DSKETCHBOOK="/Users/grant/uav/ardupilot"', '-DCONFIG_HAL_BOARD=HAL_BOARD_PX4', '-DHAVE_STD_NULLPTR_T=0', '-DAPM_BUILD_DIRECTORY=APM_BUILD_ArduCopter', '-DSKETCH="ArduCopter"', '-DSKETCHNAME="ArduCopter"', '../../libraries/AP_AccelCal/AP_AccelCal.cpp', '-c', '-o/Users/grant/uav/ardupilot/build/px4-v2/libraries/AP_AccelCal/AP_AccelCal.cpp.4.o']

My configure looks like this:
MBPro1:ardupilot grant$ ./modules/waf/waf-light configure --board px4-v2
Setting top to                           : /Users/grant/uav/ardupilot 
Setting out to                           : /Users/grant/uav/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : px4-v2 
Using toolchain                          : arm-none-eabi 
Checking for 'clang++' (C++ compiler)    : /usr/bin/clang++ 
Checking for 'clang' (C compiler)        : /usr/bin/clang 
Checking for HAVE_CMATH_ISFINITE         : no 
Checking for HAVE_CMATH_ISINF            : no 
Checking for HAVE_CMATH_ISNAN            : no 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : no 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no 
Checking for header endian.h                   : not found 
Checking for header byteswap.h                 : not found 
Checking for program 'cmake'                   : /usr/local/bin/cmake 
Checking cmake version                         : 3.6.1 
Checking for program 'ninja, ninja-build'      : not found 
Checking for program 'cp'                      : /bin/cp 
Checking for program 'python'                  : /Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python 
Checking for python version >= 2.7.0           : 2.7.11 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/local/bin/git 
Gtest                                          : PX4 boards currently don't support compiling gtest 
Checking for program 'arm-none-eabi-size'      : /usr/local/bin/arm-none-eabi-size 
Benchmarks                                     : disabled 
Unit tests                                     : disabled 
'configure' finished successfully (1.762s)

I can build copter just fine with make but its using gcc when I do that.  Is there any way to tell waf to use gcc instead of clang on OSX?  I tried the TOOLCHAIN variable and had a read of toolchain.py but I wasn't able to get it to use gcc.

#### Version
latest master

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover

OXINARF, 2016-08-05:
You can use the options `--check-c-compiler=gcc` and `--check-cxx-compiler=g++`

I intend to get Clang tested in Travis again, but unfortunatelly Clang in macOS is a custom Apple version that not always corresponds to what the Clang we test has.

gmorph, 2016-08-30:
Yep that works.  To build on OSX I did:
./modules/waf/waf-light configure --board px4-v2 --check-cxx-compiler g++
./modules/waf/waf-light build --target bin/arducopter-quad

Thanks, Grant.

closed 2016-08-30 by gmorph