ArduPilot/ardupilot #4627: Copter: Spektrum satellite failsafe feature

kovmarci86, 2016-08-06:
It would be very nice to have fail safe support when using Spektrum Satellite receiver.

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

I'm using a PixRacer on a 250 size quad, but this should be available for all types of hardware and probably for plane, rover, etc.. supporting Spektrum serial input.

kovmarci86, 2016-08-06:
Hi,

This would be my proposal:
As Spektrum Satellite receivers stop sending data in case of a signal lost, there is no way to use the throttle PWM value. To overcome this, the system may have a timeout for the receiver. When the timeout reached, may simulate a predefined throttle or 0. This will trigger the fail safe as usual.

Can you please advice where are the components most suitable for this change? I'm about to fix this.
Do you think this should be implemented around here? 
libraries/AP_HAL_Linux/RCInput

kovmarci86, 2016-08-10:
@rmackay9 : Hi, Can you please advice if you see this technically feasible?

rrr6399, 2016-08-10:
Just make sure you don't eliminate the capability to fly without a receiver at all.

kovmarci86, 2016-08-10:
Thanks, thats a good thing to keep in mind. Moreover this logic should not be triggered unless the dsm processor had meaningful data before.
Im open for more ideas/requirements.

WickedShell, 2016-08-10:
Isn't this already managed under failsafe logic? On plane at least this is
covered fairly well with the failsafe system.

kovmarci86, 2016-08-10:
As far as i understood, the data processing wont start without having available data, callbacks wont be called. I found a timer tick function, where i can check for a timeout, and set a predefined channel override. Overrides are available there as well, but i'm not what are they used for.

WickedShell, 2016-08-10:
RC overrides are for RC input via mavlink.

On plane I can assure you that RC failsafe behaves quite well with a
spektrum satellite input, I can't attest to copter either way however.

kovmarci86, 2016-08-14:
Do anyone know how this is implemented for plane? I cant find any related in the code.
As far as I know they are sharing the same RCInput files.

WickedShell, 2016-08-14:
Loss of satellite activates RC failsafe if enabled (with appopriate
timers). If disabled then the vehicle will continue in the same flight mode
as if nothing happened (unless in manual, in which case plane will change
modes to circle).

kovmarci86, 2016-08-15:
This is from RCInput.cpp for PX4 and VRBrain
> // note, we rely on the vehicle code checking new_input()
> // and a timeout for the last valid input to handle failsafe
so rc failsafe is impemented elsewhere.

For Plane, I've found the handling logic: 
> void Plane::check_long_failsafe()
in ArduPlane/system.cpp

The same does not exist for copter (as far as i know). Please correct me if im wrong.

kovmarci86, 2016-08-16:
Added an initial implementation, not tested yet, just to start a more deep discussion.
In this version, i'm using rc overrides to achieve the failsafe, but this should be changed to set an RC pwm value instead.
Also this bring up some questions.. 
Where can i reliably get:
- failsafe PWM value? Is that ok to use some default?
- throttle channel number? always 3? if this can be read from config, then how?
- do we need a PARAM to enable-disable this? when enabled, and failsafe occurs, throttle will be pulled low. without a throttle failsafe, this may be more problematic than doing nothing.

For code, see:
https://github.com/kovmarci86/ardupilot/tree/Copter-DSM-failsafe

kovmarci86, 2016-08-16:
Update: Failsafe works.
Sorry for rubbing your time. I'm not sure what did I miss, but RC failsafe using Spektrum DSM satellite works well. 
In the meantime I update the implementation i have, but will be abandoned.
Closing this.

I would be thankful if someone can point me to the existing implementation.

closed 2016-08-16 by kovmarci86