ArduPilot/ardupilot #4630: Plane: PX4 Mixer fails to program reliably leading to no working overrides

WickedShell, 2016-08-07:
#### Issue details
PX4 mixer appears to be working, but the override is not correctly configured for some reason, which leads to the overrides not working. This in turn leads to an inability to override the aircraft for safety of flight reasons or otherwise. I've sen this twice now out of 20 or so flights. And again the message of PX4IO override enabled came out, but that is simply a statustext from the FMU and is meaningless. I *strongly* move that message is either removed or only be emitted under positive confirmation that override is working, instead of just monitoring and deciding that the PWM is in the correct spot to enable overrides. This current behavior has caused multiple users to be extremely confused about the state of the overrides.

#### Version
c2da36d

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
#### Hardware type
Pixhawk

#### Logs
Not yet uploaded (need to make a smaller one, rather then uploading something near 300MB)

magicrub, 2016-08-07:
what statustext message is it?

WickedShell, 2016-08-07:
https://github.com/ArduPilot/ardupilot/blob/5a544002cbd47843058e45d167ce62544e22df5f/ArduPlane/control_modes.cpp#L79

https://github.com/ArduPilot/ardupilot/blob/5a544002cbd47843058e45d167ce62544e22df5f/ArduPlane/control_modes.cpp#L89

The FMU is frankly making the assumption here that the PWM specified is
working, which is not always a correct assumption, and when it's not
working correctly this behaviour is *extremely* misleading, which can
compromise the safety of flight as the user has been misinformed about the
status of overrides. The only indication that works at the moment is the
blue led on the IO module. Any other reporting mechanism to the ground
should be considered suspect.

WickedShell, 2016-09-06:
Fixed by #4768

closed 2016-09-06 by WickedShell