ArduPilot/ardupilot #4633: ADSB Ping heartbeat is not consumed so misconfiguration cannot be detected

rmackay9, 2016-08-08:
We do not consume any heartbeats from the uAvionix ADSB Ping sensor which means that we cannot detect a misconfiguration (i.e. incorrect baud rate).  As it is, a misconfiguration will just appear as if there are no planes nearby.

We should consume the heartbeat and then if we are not receiving a heartbeat but the user has set the ADSB_ENABLE parameter to 1 we should report an sensor health in the EXTENDED_SYSTEM_STATUS message and trigger a pre-arm check.

magicrub, 2016-08-08:
@jwuavionix would be a good person to talk to about this. I agree that adding a heartbeat is the way to go.

rmackay9, 2020-01-30:
I discussed this with @meee1 today and it seems that the Ping sensor is indeed sending a heartbeat now so we can technically implement this change now.

Pedals2Paddles, 2020-01-30:
Do all ping sensors send a heartbeat? And have they always been sending it?  Or were there older unis that don't have a heartbeat?

WickedShell, 2020-01-30:
You get heartbeats from the hardware that is integrated into the Cube carrier boards, you do not get heartbeats from the standalone PingRx (at least the one I bought direct from them a month ago doesn't do it)

Pedals2Paddles, 2020-01-30:
I was afraid of that. I don't get heartbeats from my PingRX (3 years old) either.  So looks like a new parameter such as `ADSB_HEARTBEAT` will be needed to consume it selectively.

rmackay9, 2020-01-30:
we could probably re-use the ADSB_ENABLE parameter, do our regular trick of changing it to an ADSB_TYPE parameter.

Pedals2Paddles, 2020-01-31:
That's even better.  I happen to have an ADS-B carrier on a copter now, so perhaps I'll take a stab at this one at some point.

Pedals2Paddles, 2020-02-01:
Attempting here but so far, no joy. https://github.com/ArduPilot/ardupilot/pull/13443