ArduPilot/ardupilot #4634: Unable to build Antenna Tracker due to library error

TimothyGold, 2016-08-08:
#### Issue details
I have been able to build the cloned repository fine for a long while but I ran into an error that showed up after some library changes were made.  The error I get is below.

In file included from c:/Users/Tim/Documents/GitHub/ardupilot/libraries/AP_HAL/utility/sparse-endian.h:23:0,
                 from c:/Users/Tim/Documents/GitHub/ardupilot/libraries/AP_ADC/AP_ADC_ADS1115.cpp:2:
c:/Users/Tim/Documents/GitHub/ardupilot/libraries/AP_Common/missing/byteswap.h: In function 'uint16_t __bswap_16(uint16_t)':
c:/Users/Tim/Documents/GitHub/ardupilot/libraries/AP_Common/missing/byteswap.h:11:42: error: '__builtin_bswap16' was not declared in this scope
compilation terminated due to -Wfatal-errors.
make[3]: *** [libraries/AP_ADC/AP_ADC_ADS1115.cpp.o] Error 1

#### Platform
[  ] All
[ X] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

OXINARF, 2016-08-08:
This is a support question and those should be posted at the forum (http://discuss.ardupilot.org) or, being related to development, at the Gitter channel (http://gitter.im/ArduPilot/ardupilot).

Anyway, you need to update your Windows toolchain.

closed 2016-08-08 by OXINARF

rmackay9, 2016-08-09:
You can find the toolchain here (look for pixhawk_toolchain_installer_latest.exe):
http://firmware.ardupilot.org/Tools/PX4-tools/

Please note that it installs in a different directory (it's C:\Pixhawk_Toolchain instead of C:\px4) so you may need to update some shortcuts (if you created any).