ArduPilot/ardupilot #4655: Copter: ESC Calibration with ArduPilot - OneShot+OneShot125

lvale, 2016-08-12:
#### Issue details
ESC calibration
#### Version
3.4rc2
#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
quad
#### Hardware type
PixHawk, PixRacer
#### Logs

We've had various reports of people not being able to calibrate ESC's when having OneShot or Oneshot 125 selected.

Well today I was finishing up a new build of a race quad and it happenned to me too. Tried every possible combination but only managed to calibrate the ESC's if the ESC's were defined as normal.

lvale, 2016-08-14:
As a follow-up:

Initial tests were performed with BLHeli fw version 13.2

Updated the ESC's to the latest BLHeli fw 14.6 and repeated tests with the same results, ie no calibration achieved under Oneshot/Oneshot125

tabascoz, 2016-08-17:
Hello, I have hk afro esc running blheli 14.6 and here is my results: 

RC1: 
pwm_type=1 - Works OK,  including calibration
pwm_type=2 - Can't calibrate or arm esc

RC2: 
pwm_type=1 - Works with issues. Can calibrate, but esc delays arming and it seems to be a random time. 
pwm_type=2 - Can't calibrate or arm esc

rmackay9, 2016-08-29:
Re the delay in arming with -rc2.  Just to avoid some confusion, -rc2 includes a 2 second delay after arming before the motors start spinning.  This gives some time for the buzzer to sound (assuming you have one) before the motors start spinning.

@lvale, @tabascoz,
I'm investigating this today.  Were you using a safety switch?

rmackay9, 2016-08-29:
@tridge and I investigated this today and found/fixed the issue.  Here's the exact patch from Tridge and the auto build server should create a new binary within a few hours.  https://github.com/ArduPilot/ardupilot/commit/d186e1c6484d90ce294977beebc045e03876fe88

Could someone (maybe @lvale) test this?  I'm sure you know how to load up master onto your vehicle but for anyone else, the MP's Install Firmware screen has a magic keystroke (Ctrl-Q) that allows you to load "latest" onto your board.  So if you load "latest" at least 3 hours after this post, you should get a version with the fix.

By the way, I hope to push out -rc3 this week.

rmackay9, 2016-08-29:
I'm going to close this item but we can re-open it if we find the fix doesn't work.

closed 2016-08-29 by rmackay9