ArduPilot/ardupilot #4661: Please deny operation if SD not installed (make Arducopter, like Plane)

AndKe, 2016-08-13:
Current Arducopter release is willing to boot, operate even if no SD for logging is installed.

Today - I lost a whole day of work, last file in /APM/LOG was from yesterday - LASTLOG.TXT also pointed only to last log of yesterday.

I am not actually *sure* whatever the SD was not detected, or some fault with it's filesystem just make Pixhawk fail to create files for todays >10 flights.

I did backup the SD card using gnome disk utility - before fixing errors , but fixing errors, or deleting it's files did not help. 
Finally, when I formatted the partition on the SD, then AP was able to log just fine.

The SD backup can be uploaded within few days (if you want to reproduce),  it's too big to upload from here.  The interesting part is that no amount of file deletion, and error-checks will make this partition work again - it seems - only format.   While it mounts and works as exepcted anywhere but pixhawk.

R-Lefebvre, 2016-08-13:
This would have to be an option.  Sometimes I remove an SD card to put it in a computer to get log files.  Then forget to replace it, and go for a flight.  I like that I can fly without driving home to get a card.

AndKe, 2016-08-13:
yes, option is sufficient, like in arduplane ARMING_CHECK bit 10 determines whatever to allow arming or not if there is no logging.

I understand your need, but at the same time, for me flying w/o logging is a waste of time, and - even worse; I specified that AP is logging in my operation manual ,(approved by Norwegian CAA). So by taking off without logging (due to error), I am unknowingly violating my own operating procedures.

peterbarker, 2016-08-17:
If we can get https://github.com/ArduPilot/ardupilot/pull/4677 in, using logging_checks from AP_Arming_Copter will be trivial.

peterbarker, 2016-08-17:
@AndKe Also note that we've just started to include logging as a mavlink sensor, so you can tell from your GCS logging is good.  Well, assuming your GCS knows about it :-)

rmackay9, 2016-09-01:
I've added an issue to the MP issues list to report the dataflash logging health because @peterbarker has added the reporting to the system status message. https://github.com/ArduPilot/MissionPlanner/issues/1352

AndKe, 2016-09-01:
Do we know whatever the sys status bit will reveal a problem where filesystem apparently is OK, but a new log file failed to be created ?
If yes, this is a great solution, combined with optional arming check(that checks it)

peterbarker, 2016-09-01:
@AndKe you will only know if you can create a file when you... go to create a file :-)

If your logfile is open, the sensor is "OK".  If we're not currently trying to log to a file the sensor is disabled.  If you don't have logging enabled, it is "not present".  If logging has failed then the sensor indicates failure.

@rmackay9 I've already raised a PR for MAVProxy :-) https://github.com/ArduPilot/MAVProxy/pull/297

Sulter, 2016-11-01:
I would like to add that this is also a feature that would help us a great deal. We already had numerous episodes where we ended up without logs, due to ardupilot simply silently changing the logmask to 0 after failing to initialize/mount.

AndKe, 2016-11-02:
Is it already an option for arming check denies arming if logging failed ?  (it was mentioned earlier)

peterbarker, 2017-12-19:
@Sulter we definitely no longer set `LOG_BITMASK` to zero.

@AndKe yes, we have log checking for arming.

closed 2017-12-19 by peterbarker