ArduPilot/ardupilot #4678: AP_Arming: Primary compass misconfiguration may cause use without prearm checks

peterbarker, 2016-08-17:
#### Issue details
If the primary compass is not enabled then we skip compass checks: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Arming/AP_Arming.cpp#L262

The primary compass is simply nominated with a parameter: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass.h#L274

A compass instance may be nominated to be used by NavEKF: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_Control.cpp#L320

Taken together, the compass instance used by EKF may not have been tested at all for consistency.

@tridge suggested that the primary instance should probably be dynamic (as the equivalents in other libraries are), and that if the current primary goes unhealthy (or marked as not-to-be-used) then another instance might take its place.

#### Version
master
#### Platform
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Any

#### Hardware type
n/a

#### Logs
n/a

Naterater, 2019-01-08:
Is this still a problem @peterbarker? Considering the CAN Compasses that are coming as compasses with higher numbers, this could become more important.

peterbarker, 2019-01-09:
I think @WickedShell may have addressed this problem.

closed 2021-01-21 by peterbarker