ArduPilot/ardupilot #4679: AHRS's getLastYawResetAngle provides the incorrect yaw angle change if primary EKF is changed

rmackay9, 2016-08-17:
This issue was discovered as part of Copter-3.4 beta testing.
http://discuss.ardupilot.org/t/uncommanded-yaw-movement-rc2/10780/2

Many EV, 62 messages are shown in the dataflash log as the AHRS switches between the two EKFs.  Upon further investigation we think the yaw-angle-reset feature is not being provided with the correct yaw angle correction.

This is the location in the Copter vehicle code that consumes the EKF's yaw-angle-change.
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/Attitude.cpp#L56

priseborough, 2016-08-25:
I've created a  fix and will submit it in a PR when I have had a chance to test it:

https://github.com/priseborough/ardupilot/tree/pr-ekf2YawResetPublish

rmackay9, 2016-09-14:
This has been merged into master and will go out with -rc5 so closing this issue.

closed 2016-09-14 by rmackay9