ArduPilot/ardupilot #4693: Enter Barometric pressure adjustments (i.e. 29.92, 30.08, etc)

uavpilotjoe1981, 2016-08-18:
#### Issue details
Method of entering the barometric pressure setting (i.e. 29.92, 30.08, etc)

#### Version
None of the versions have this capability

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
_Planes_

#### Hardware type
_PixHawk_

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2016-08-18:
There Ground pressure is recorded and available via [GND_ABS_PRESS](http://ardupilot.org/plane/docs/parameters.html#gnd-abs-press-absolute-pressure) which is in Pascals. A companion computer could use that to compute an offset in meters to set GND_ALT_OFFSET, but having a direct pressure param to write to would be nice too

magicrub, 2016-08-18:
Here's a conversation from @uavpilotjoe1981 from diydrones posts starting here:
http://diydrones.com/profiles/blogs/apm-plane-3-6-0-released?xg_source=msg_com_blogpost&id=705844%3ABlogPost%3A2266847&page=4#comments

 

> If you are flying at 3000' and a maned airplane is flying at 3500' indicated and the Barometric pressure changes and the manned airplane adjusts his Alt. setting. He is now too high and will be lowering alt to meet up with the desired alt of 3500' The UAS does not make this adjustment and is still at the 3000 but actually it is at 3500.. See where i am going with this..... They are reporting different alt. but at the same alt. This is going to be an issue for long endurance flights. We need an easy way to enter new altimeter settings like manned aircraft so that we are also adjusting for barometric changes.
> 
> What about when a flight will take place in Class A airspace. At these altitudes the pilot is required to input standard pressure 29.92. How are we to handle this? 
> I ask because I work at the NMSU UAS Flight Test Center and we have a proponent that is using a Pixhawk AP and they want to eventually go to 60k ft. The FAA and We are concerned with the fact that we cannot adjust the Barometric preasure setting to match all other aircraft operating in the airspace.

I've updated the original message to include this info.

WickedShell, 2019-06-27:
You can directly manipulate the GND_ABS_PRESS in pascals now. This must be referenced against your home altitude, rather then sea level however. If you assume standard lapse rate and know the home position then this can be directly calculated. (Caution: large step changes to this can upset the EKF)

closed 2019-06-27 by WickedShell