ArduPilot/ardupilot #4697: Copter-3.4: terrain following may lead to endless climb if sonar max alt is configured incorrectly

rmackay9, 2016-08-19:
Tridge has found an issue in SITL in which the vehicle kept climbing and climbing during AUTO when it was requested to fly to some altitude (say 10m) but the range finder (whose maximum range was 50m) and the range finder didn't report that it went out of range.  It's response was as if it was seeing a good value of 50m.
This situation can arise especially with analog range finders if the user simply sets the maximum range too high.
We should think of ways to reduce the change of this happening.
Idea #1: use google earth data if the waypoint mission altitude-above-terrain is obviously higher than the range finder could ever measure (i.e. > rangefinder.max_cm).

tomlauzon, 2016-08-19:
Here's an idea: when within range (i.e distance measured is not constant) calculate the difference between EKF altitude and rangefinder, as a reference. If this difference changes beyond a threshold (e.g. 3m) assume that the rangefinder limit has been reached and don't use rangefinder anymore. Recalculate the reference difference every minute or so because of baro drift during longer flights.

chris669, 2018-02-28:
should be mentioned somewhere in documentation, unless it will really be corrected in 3.6.0 and not removed from milestone. As RTL does not work due to this, it should be set as a high priority,just my point of view. Chris

chris669, 2018-03-09:
So the issue has been closed? Many other unexpected crashs will happend

chris669, 2018-04-27:
any news about this issue? thanks.

auturgy, 2018-04-27:
I made a couple of wiki changes but not merged yet.

chris669, 2018-05-08:
I know this is a complex point, but confusing sometimes. I was told to modify an EKF parameter EK2_ALT_SOURCE=1 http://ardupilot.org/plane/docs/common-apm-navigation-extended-kalman-filter-overview.html#commonly-modified-parameters but after i read in another part of the wiki that: Setting 1 uses the range finder and is only available in combination with optical flow navigation (EK2_GPS_TYPE = 3)
So I'm not a genius and there could be a high probability that people like me would miss a point. 
A Matrix describing the implied parameters and their compatibilities with other ones could be a solution?
Or a list of standard parameters that could be used without high risk of fly away? 
For example in the Range Finder page, there is no instructions for the TFMINI lidar, there is a lot comments on the forums about how to right configure this lidar, but nothing in the Wiki. Thanks for reading.

rmackay9, 2018-05-09:
@chris669, I'm always confused when I hear that people have modified the EK2_ALT_SOURCE.  IT's really not something that the vast majority should change.

The TFMini lidar wiki page is coming soon.  Benewake has just joined as a partner and I must admit that I had never heard of this lidar before then.  Others in the team had but I hadn't and I probably do 80% of the wiki updates.  Anyway.. coming soon.

amilcarlucas, 2021-03-09:
This is AFAIK been fixed in Copter 4.0.7. Can you test that?
Ad master has an even better fix.

rmackay9, 2021-03-10:
@amilcarlucas I think this issue still exists in master.  I'm not sure there actually is a solution when using analog sonar.