ArduPilot/ardupilot #4701: Arming check for mission sanity-check

magicrub, 2016-08-19:
Add an Arming check that checks for things like:
- takeoff alt that is underground (accidental absolute alt)
- negative takeoff pitch

some items should be allowed but produce a STATUSTEXT warning such as:
- land point alt is not 0
- any waypoint is under-ground. This is OK if launching off a cliff and you're flying down, we don't want to prevent that.. right? or maybe we do and the mission check needs to be disabled for that corner case?

WickedShell, 2016-08-19:
I think flying below ground is enough of a corner case that you should
probably need to disable the check. If they're is opposition I could see it
as an extra ARMING_MIS_STRICT that defaults to disabled, but it set to 1
checks the altitudes for not being below ground.

Re landing waypoint altitude, I'm working on a pr that uses landing to do
deep stalls where the altitude is the entry altitude, so this would be a
problem for me. I could probably just add a new MISSION_CMD though for deep
stall as a workaround.

charleslinquist, 2016-09-30:
MP should warn for route lengths greater than set length and missions that don't include a LAND command. 

Also, the battery mAH counter in APM gets reset every power cycle. I watch the percentage of battery left much more than the battery voltage, because with percentage, I don't have to do any mental arithmetic.  I have mistakenly connected a battery that wasn't fully charged, and started a mission.  I looked down and the remaining battery percentage was 80%, but the battery voltage was at the discharge level.

I think that the "remaining" percentage should have a voltage component to prevent this.

R-Lefebvre, 2016-09-30:
The battery voltage thing has been talked about many times.  It is basically possible to do with any reasonable certainty.

charleslinquist, 2016-09-30:
>
> The battery voltage thing has been talked about many times. It is 
> basically possible to do with any reasonable certainty.

I have had much better results with my battery voltage readings since I 
modified AP BATT_MONITOR.CPP as follows:

     if(_monitoring == AP_BATT_MONITOR_VOLTAGE_ONLY || _monitoring == 
AP_BATT_MONITOR_VOLTAGE_AND_CURRENT) {
         // this copes with changing the pin at runtime
         _volt_pin_analog_source->set_pin(_volt_pin);
         _voltage = (_volt_pin_analog_source->voltage_average() * 
_volt_multiplier)+ (_current_amps * _batt_esr);
         if(_volt2_pin_analog_source != NULL) {
             _voltage2 = _volt2_pin_analog_source->voltage_average() * 
_volt2_multiplier + (_current_amps * _batt2_esr);     ;
         }
     }

The addition of the _battx_esr   (which I presently have set to .016f) 
term  compensates nicely for the ESR voltage drop of my battery
under load. I measured my battery ESR + drop in the wires leading to the 
voltage sensor, and found that the resistance is about 16 milliohms.

With that addition, I find that my displayed battery voltage is close to 
constant, regardless of the current draw.  During flights, I get a nice 
downward slope of
battery voltage regardless of whether I'm climbing or cruising or - 
whatever.  Of course, the slope of the line is dependent on the draw, 
but it no longer
has big "steps".

With my change, I'm measuring the actual internal CELL voltage, as 
opposed to TERMINAL voltage.  My only issue now is that my ESR value is 
fixed in the firmware.
Yes, I have variable names, but the values are equated to static values.
I don't know how to add a new parameter to Mission Planner and Mavlink 
in order make a variable that I can change as my battery ages.  If you 
could help me there,
I would appreciate it.

I was thinking of making this process automatic by measuring the 
(uncompensated) voltage and current at a low-current point(like a hover) 
and again during a climb.
If the two measurements were taken only a few seconds apart, the ESR 
could be calculated by   (Voltage(hover) - Voltage(climb)) / 
(Current(climb) - Current(hover))

I should also mention that I added a more effective averager for the 
_current_amps variable.  Now my current readings are a bit smoother than 
before.  This also
helps with my modified battery voltage routine.

Charles Linquist

magicrub, 2016-09-30:
@charleslinquist please don't hijack the thread. This is about arming check. Open a new issue for a completely different topic.

magicrub, 2016-09-30:
moved your thread here: https://github.com/ArduPilot/ardupilot/issues/4911 and will be deleting the posts here as to not side track it too much.

charleslinquist, 2016-09-30:
Sorry

magicrub, 2016-12-03:
I plan to implement this soon. Any last minute requests? We can, of course, add more later too.

WickedShell, 2016-12-03:
Land point alt not 0 should be a flag/possible to disable, as there are
normal missions that can be flown that rely on that behavior and a user
shouldn't have to see a warning if it's intentional.

Sanity check that all jump targets in the mission exist.

Sanity check that if rally points are planned that the planned flight path
will be in range of at least a really point. (This one is potentially quite
computationally intensive, in fine with saying this should be managed by
the gcs.

Loiter radiuses that are unachievable based on bank/airspeed settings.

Loiter radiuses that have exceeded the maximum possible storage (this is
also a generic AP_Mission problem. It needs to intelligently clamp the
radius and earn if it did that)

Endless jump sequence with no nav waypoints inside it.

Landing start maybe wapoint, but no actual nav land waypoint.

Unachievable glide slopes (ie running says the aircraft can't have done the
decent from WP 4 to work 5 because the slope exceeds the limits in the
tuning) this should maybe be optional, but it would be a nice way to let
users know they planned the impossible.

magicrub, 2016-12-03:
Several of those seem reasonably easy to implement. The hard ones I'll push for next round.

magicrub, 2017-08-29:
also check if first waypoint is very far away from home. This is common when you boot up in auto and launch using an old mission designed for a different location

juvinski, 2017-08-29:
Hi @magicrub,
maybe - a altitude takeoff complete to - I'm using qground and now there is a feature to change altitude for all elements on the mission - including takeoff - and how the takeoff complete altitude can be a value around 40 or 50, if the takeoff near to 100 mts an alert can be triggered ?

magicrub, 2017-08-29:
@juvinski I think I'm missing a detail. A takeoff of 100m is OK. No alert should be given for that.

magicrub, 2017-09-04:
Good example of where this would come in handy
https://www.youtube.com/watch?v=lwazVDZwrK4

juvinski, 2017-09-04:
Hi @magicrub,

sorry for the delay. I know the 100 meters is possible but in my point of view 100 meters is excessive once the mission will not start before takeoff finished. It's just my opinion

magicrub, 2017-09-04:
What if the mission is takeoff=1000m? That could be OK in certain scenarios. This needs to be a generic check. Perhaps a better check is  (takeoff.alt <= firstNavWaypoint.alt*1.5)
I can't think of any good reason to have the takeoff drastically higher than the next NAV_WAYPOINT point. Can you?

juvinski, 2017-09-04:
I think would be perfect.

magicrub, 2017-10-26:
another one: inhibit NAV_TAKEOFF takeoffs when VTOL is configured, we expect to be NAV_VTOL_TAKEOFF . Same goes for NAV_LAND and NAV_VTOL_LAND. @tridge wants to allow NAV_TAKEOFF because there are rare instances where a VTOL aircraft could indeed do a "fixed wing" horizontal takeoff but that can be handled a different way with some Q_* params

peterbarker, 2017-12-19:
@magicrub Anything you can get into a PR for this?

magicrub, 2018-01-09:
added https://github.com/ArduPilot/ardupilot/issues/1348 to the list at the top