ArduPilot/ardupilot #4702: Copter: parachute release is separate from crash-check allows possibility of parachute not being deployed in crash

rmackay9, 2016-08-19:
The code for the regular crash-check is quite separate from the parachute release check.

They use almost exactly the same logic except that the parachute triggers after 1sec of loss of control while the crash-check triggers after 2 seconds.  Also the parachute-check checks that the barometer says we're falling.

The separation of the logic means we're open to the possibility of the crash-check triggering without the chute deploying.  Maybe that should be possible because we could imagine a simple flip on take-off (we might not want the parachute to deploy in this case).. but perhaps we should at least combine the logic and possibly delay the crash-check timeout if a parachute is installed (to give time to see if the vehicle is falling).

rmackay9, 2016-09-06:
This has been partially resolved in that the parachute should deploy more easily now reducing the chance that the crash detector will fire first.
https://github.com/ArduPilot/ardupilot/commit/1f2d17e8e78f556e1a432940c05f975d5c71db4c.
I'm going to move this issue out of the AC-3.4 milestone because I think the above fix is sufficient.

Naterater, 2019-01-08:
Is the fix sufficient enough to close this issue?

rmackay9, 2019-01-09:
@Naterater, i'd forgotten about this.. i think we should probably leave it open as a reminder..

peterbarker, 2019-01-09:
Wow.  Talk about timing.
   
I hit this problem just now when writing
https://github.com/ArduPilot/ardupilot/pull/10208

Naterater, 2019-01-17:
@peterbarker, since that PR is merged, does that solve everything here?

rmackay9, 2019-01-18:
@Naterater, the autotest is there now and highlights the issue.. but the underlying issue itself has not been fixed

peterbarker, 2019-12-24:
`./Tools/autotest/autotest.py build.ArduCopter fly.ArduCopter.Parachute --gdb --debug`

tridge, 2019-12-24:
related to this, should we fire chute on armed watchdog?

peterbarker, 2019-12-24:
```
STABILIZE> param fetch SIM_ENGINE_FAIL
STABILIZE> Requested parameter SIM_ENGINE_FAIL
SIM_ENGINE_FAIL = 1.0000000
height 30
height 20
APM: Crash: Disarming
APM: Disarming motors
DISARMED
height 10
APM: PreArm: DCM Roll/Pitch inconsistent by 26 deg
Timed out looking for BANG
AUTOTEST: Exception caught: Traceback (most recent call last):
  File "/home/pbarker/rc/ardupilot/Tools/autotest/common.py", line 2232, in run_one_test
    test_function()
  File "/home/pbarker/rc/ardupilot/Tools/autotest/arducopter.py", line 2275, in test_parachute
    self.mavproxy.expect('BANG')
  File "/home/pbarker/rc/ardupilot/Tools/autotest/pysim/util.py", line 406, in _expect_callback
    raise pexpect.TIMEOUT(timeout)
TIMEOUT: 60
```

ddomit, 2021-08-28:
I think this issue is still present! Im doing the tests by arming in stabilize, then rotating the quad more than 20 degrees and i get a crash and disarming motors but not releasing parachute!

Im in beta 4.1 and manual tests with the remote do release the parachute.

Here is the log https://drive.google.com/file/d/1Eha9vKNj-sIjxJ-pdNBw6bcIsmEnjLZX/view?usp=sharing

ddomit, 2021-08-28:
Maybe Im missing something or the documentation (here https://ardupilot.org/copter/docs/parachute.html) is outdated, how can i test the automatic deployment of chute?