ArduPilot/ardupilot #4717: frame reference clarification

khancyr, 2016-08-23:
#### Issue details
My question is about difference about local_ned and body_ned.
We got to definition : 
https://github.com/ArduPilot/ardupilot/pull/4716 
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1717

And the wiki state another : 
http://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html#copter-commands-in-guided-mode-set-position-target-local-ned

For me MAV_FRAME_BODY_NED is wrong in wiki : should be 
````
Positions are relative to the vehicle’s home position in a frame based on the vehicle’s current heading
````
instead of 
````
Positions are relative to the vehicle’s home position in the North, East, Down (NED) frame.
````

Then for MAV_FRAME_BODY_NED we should not apply offset correction from current pose but home pose.

What do you think ?

OXINARF, 2017-06-04:
I agree with @khancyr that BODY_NED definition on the wiki (and the way we handle it in Copter) seems wrong to me and should instead be what @khancyr wrote, relative to home in a frame based on vehicle's heading.
In my opinion, MAVLink definitions are terrible and don't help here at all, but maybe that's just me.

IamPete1, 2021-08-30:
Sounds like this is a wiki issue?

rmackay9, 2021-08-31:
I agree this is a wiki issue so I'll close this.  I hope that in the near-ish future I will find the time to update this relevant wiki page: https://ardupilot.org/dev/docs/mavlink-commands.html

closed 2021-08-31 by rmackay9