ArduPilot/ardupilot #4722: SITL Arm-to-Takeoff Delay

jmodares, 2016-08-25:
#### Issue details
_Please describe the problem, or desired feature_
We performed a 5 round test to see how accurate SITL is, and we found that there is about 1.5 seconds delay between SITL and actual drone. The mission is simple, (1) take-off to 5 meters (2) travel 5 meters to east (3) land. The Arm-to-takeoff for SITL is around 4.5 seconds, but for actual drone is about 3 seconds.

#### Version
_What version was the issue encountered with_
Copter 3.4

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad, H shape

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk

#### Logs
_Please provide a link to any relevant logs that show the issue_
[delay.pdf](https://github.com/ArduPilot/ardupilot/files/436179/delay.pdf)

khancyr, 2016-08-25:
I don't know if it is really an issue.
On my hexa, it can take more than 5s to takeoff.... I use an automatique script to trigger takeoff with 5s timeout and I got plenty of timeout. It could depend of meteo condition etc

magicrub, 2016-08-25:
Do all params match on sim vs real aircraft? Also, the physics engine is
not perfect. This will most likely be closed with will-not-fix.

jmodares, 2016-08-26:
Yes, all parameters are the same. I'm not sure if arm-to-takeoff depends on the model or frame. I plot the average longitude and altitude change of 5 round experimental results and simulations with a 1.5 second time shift.
[lon.pdf](https://github.com/ArduPilot/ardupilot/files/439186/lon.pdf)
[alt.pdf](https://github.com/ArduPilot/ardupilot/files/439185/alt.pdf)