ArduPilot/ardupilot #4729: Porting to OrangePi

xtr0d3m0n, 2016-08-26:
#### Issue details
any body working on porting over the HAL to this cheap opensource SBC?It more power full than Raspbery pi and  1/3 of the cost.if you guys want to do some coding I will be happy to do testing I can easily rig up some IMU breakout boards etc and its pin compatible with Rpi so I guess u can use existing hats?.  http://www.orangepi.org/

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

cudnn, 2016-08-26:
i have port to banana pi.  a20 cpu

xtr0d3m0n, 2016-08-26:
@cudnn AWESOME !! link please

magicrub, 2016-08-26:
If it's Rpi compatible, what if anything needs to change? Does installing
the RPi shield and loading the RPi code work?

xtr0d3m0n, 2016-08-28:
I ordered a board ..let see

lucasdemarchi, 2017-04-21:
> If it's Rpi compatible, what if anything needs to change? Does installing
> the RPi shield and loading the RPi code work?

Except for RCInput and some other things that are rpi-specific (actually SoC-specific), yes.

closed 2020-11-15 by xtr0d3m0n