ArduPilot/ardupilot #4731: EASA requierements for certified UAS

severinleuenberger, 2016-08-26:
Hello dev-team

Are there already plans to meet / fulfill the requirements from EASA?
Because without a CE certified UAS or quite some effort to get it certified  there will be no legal flying of any UAS anymore :-(
And EASA could be already go live in 2017! So dont keep this down low.

Regarding technical requirements, I especially think about to restrict the aircraft to a specific height limit like 50 or 150m above ground  And the requirement to restrict the Aircraft to prefedined no fly zones, that cannot be changed manually (not deactivatable by users). Ssee I.6.a Geofencing class 2 and I.6.b Geofencing class 3.
Please read the regulations carefully: 
[UAS Prototype Regulation final.pdf](https://github.com/ArduPilot/ardupilot/files/439555/UAS.Prototype.Regulation.final.pdf)
[Explanatory Note for the UAS Prototype regulation final.pdf](https://github.com/ArduPilot/ardupilot/files/439556/Explanatory.Note.for.the.UAS.Prototype.regulation.final.pdf)

Best wishes from the 
SFCD (swiss federation of civil drones)

Severin 
(an enthusiastic Ardupilot User and Drone Pilot)

R-Lefebvre, 2016-08-26:
Hi Severin,

I don't think this has been discussed before.  Is that the only requirement we would have to meet, the height and geofence restrictions?  Is it a simple matter of possible a special firmware upload for EASA users?  Or would this need to be an absolute restriction in code that would affect all users around the world?  (ie: one firmware for all).  If it's just a special load for EASA users, I think that's possible.  If they expect the security of knowing that no firmware is available which breaks the requirements, requiring only one firmware with this restriction for all, then I don't think we would do that.

AFAIK, there has been no work towards meeting this.

rmackay9, 2016-08-27:
Very interesting.
As a development group, we couldn't take on the effort of getting the certification for the software ourselves but we could assist with the functional changes required to meet the specs.

From a software point of view it doesn't really sound that hard on the surface of it anyway.
- remove the ability to turn off the geofence (1hour :-))
- ensure GPS lock before take-off no matter what mode (this ensure the fence is enforced).

@severinleuenberger would you want to be the "user rep" for such a project?  I.e. you'd try to actually certify one of your drones under this new standard?  I fear that if we're missing a user who actually wants to do this it won't happen easily.

squilter, 2016-08-27:
Thanks for bringing this up. I had only been following FAA rules, and it is interesting to see how these rules are different. I have been interested in implementing geofencing into ardupilot. I think the discussion has two distinct parts:

1. Non-technical
Are we actually enforcing laws, or just informing the user about the laws? Should the user be able to override the checks? An important point here is that we don't want to frustrate our users and have them move to a different platform.

2. Technical
As Randy pointed out, the altitude restriction is not technically challenging. The geofencing is more challenging. The EASA rules talk about up-to-date geofencing, so a system is required to update the geofence data. Fortunately, I have already put some thought into this. See [this document](https://docs.google.com/document/d/14j6oO0IH-jobzfKbocxTKpZ5DFexVcmsRoLfbSQg0DI/edit?usp=sharing) and [this](http://discuss.dronecode.org/t/no-fly-zones/243) updated suggestion.

Also see #1056

severinleuenberger, 2016-08-28:
Regarding the geofencing, there will be a centralized solution, where the fence-definition will be updated. There is nothing defined yet. But there will be soon, otherwhise the developers cannot plan and build solutions market available in 2017. We are in close contact with FOCA (https://www.bazl.admin.ch/bazl/en/home.html ) and will get more infomations from there. Other big developers like DJI must change their products as well and get it certified, if they plan to sell to Europe. 

I will ask the board and our member from Swiss Federation of Civil Drones, if we get a group together, who plan to get an UAS certified. Of course I would support you here from the requirements point of view, when we will certifiy a plane or a copter. And ideally get a path to multiply the certification on other UAS.

hiro2233, 2016-08-28:
Hi @severinleuenberger, i'm interested to get and introduce UAS certification and international regulations in my country, Bolivia. There is a DGAC associated with ICAO, but there isn't a regulation for specific UAS right now. I'm a developer APM and i target Ardupilot system to be registered under the international regulation for commercial UAV use with local DGAC.

Please, tell me, what i need to become a legal certified agent?, are you able to give this certification for Bolivia?

Regarding geofencing and altitude, i think the best practice for integration are giving two options, the first, a firmware EASA as @R-Lefebvre comments and the second an external hardware integration, this one not limit the development for APM scene and continue grow up.

lvale, 2016-11-06:
Tentative targeting for 3.5, because the EASA proposals are targeted for 2017

R-Lefebvre, 2016-11-13:
Not sure if this is the best place for this, but maybe...

I've seen it suggested that the EASA regulations will also require UAV flight software to conform to DO-178B.  Is there any truth to that?

lvale, 2016-11-13:
@R-Lefebvre Not in the current proposal, but from the EU representatives I met I got the impression that for the larger and commercial UAVs that (and others) requirements might be needed to get an airworthiness certificate

R-Lefebvre, 2016-11-13:
Ok, thanks.  Somebody on RCG suggested that *no* drones larger than 250g will be legal to fly for any purpose unless they meet DO-178B, in both Europe and the US.  So just fearmongering?  The person who stated it has a history of trolling, but I just wanted to check.

lvale, 2016-11-14:
@R-Lefebvre Then you also have DO254 and being even more picky they could also go with ARINC825 .... 
:)

aercamti, 2016-11-14:
As far as timing goes, the "Prototype" Regulations currently state:
Article 14 Applicability
1. As from [2 years after entry into force of this Regulation — estimate 2019], economic operators and UASs placed on the market shall comply with this Regulation.
2. UASs placed on the market before [2 years after entry into force of this Regulation — 2019] and having a mass of 250 g or less, including payload, are deemed to be classified as class 0 as defined in Appendix I.2 and can continue to be operated according to operational subcategory A0 defined in UAS.OPEN.60.
3. As from [3 years after entry into force of this Regulation — estimate 2020], all UA shall be operated in accordance with this Regulation.
...

aercamti, 2016-11-14:
I am glad that these issues are brought up here. (Not because I like it, but because it _will_ be something that we'll need to deal with...). 
Here are some brief excerpt from the EU Prototype regulations which seem relevant (for those who do not care to wade through all 72 pages).

The expected product requirements are outlined from page 29 onward of this document: https://www.easa.europa.eu/system/files/dfu/UAS%20Prototype%20Regulation%20final.pdf 

Per appendix I.4 (e) a system (Class 2) needs to include (excerpt): 

- (1) an ‘auto-return home’ function;
- (2) a maximum attainable altitude of 50 m above the ground through performance limitations or an active limitation system;
- (3) a geofencing system as per Appendix I.6.a;
- (4) an electronic identification system as per Appendix I.6.c;

P. 31 has Appendix I.5 - excerpt:

- (1) an ‘auto-return home’ function;
- (2) a geofencing system as per Appendix I.6.b;
- (3) a maximum attainable altitude of 150 m of height from the ground, through performance limitations or an active limitation system;
- (4) electronic identification as per Appendix I.6.c;

The details of the geofencing requirements referenced above are on p. 33 of the document linked above. 
Hope this helps.

IamPete1, 2021-08-30:
I suspect the regulations have changed significantly, maybe we should close this  and open a fresh issue with the latest info.

aercamti, 2021-08-31:
Ok, fair enough. Better to make sure that any pending adaptation match the current regs, which spell out details and are in force.

closed 2021-08-31 by IamPete1