ArduPilot/ardupilot #4744: Serialx - Option 10 - Fills all the 32 telemetry slots on Taranis

lvale, 2016-08-29:
#### Issue details
After selecting Option 10 for Serialx - and going to the radio to discover the available sensors, the sensors repeat themselves and quickly fill all the Taranis slots

#### Version
Copter Master from 29-8-2016

#### Platform
[X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A
#### Hardware type
Pixracer + Taranis

#### Logs
[image]

[image]
[image]
[image]

magicrub, 2016-08-29:
@tomlauzon Please take a look at this ASAP, seems to be a critical bug

tomlauzon, 2016-08-29:
Ok, we're on it

tomlauzon, 2016-08-29:
@lvale, we just tried on our Pixracer with the same firmware you used for both Pixracer and Taranis and were not able to replicate. Everything works fine.
Can you tell us if you have other FrSky sensors connected?
Can you send us your ArduPilot parameter file?
Any special features enabled? High rate logging or high rate telemetry...? 
Do you have an original Pixracer or a clone?
Thanks for identifying this issue. We'll work on fixing it as soon as we can identify what's causing it.

lvale, 2016-08-29:
[racer.param.zip](https://github.com/ArduPilot/ardupilot/files/443230/racer.param.zip)
and it's an original PixRacer. Are there any others ? :)
No FrSky sensors connected- straight from the Racer to the X4R
Just tested with a X9E and the results are the same.
Serial4 reverted back to mode 4 - SPort on the param above

lvale, 2016-08-29:
[testquad250.eepe.zip](https://github.com/ArduPilot/ardupilot/files/443239/testquad250.eepe.zip)
adding eepe

tomlauzon, 2016-08-29:
alright, thanks!

tomlauzon, 2016-08-29:
racer.param.zip doesn't contain any parameters. Could you double check? Obrigado

lvale, 2016-08-29:
guess I'll have to open an issue :) with @billbonney because RC4 didn't save the file, but here goes

[testracer.param.zip](https://github.com/ArduPilot/ardupilot/files/443391/testracer.param.zip)

tomlauzon, 2016-08-29:
hmmm... we tried your config file but it's working fine on our side. Do you have other equipments on your other serial ports? I see all your other ports are also configured for MAVlink.

lvale, 2016-08-29:
On other machines I have several equipment connected to the serials and there is no issue with the streams of each of the separate serial ports.

Also with serial set to 4 (didn't test 3) there's no issue with the multiplication of the sensors flooding the OpenTx slots.

SERIAL0_BAUD,115
SERIAL0_PROTOCOL,1
SERIAL1_BAUD,57
SERIAL1_PROTOCOL,1
SERIAL2_BAUD,57
SERIAL2_PROTOCOL,1
SERIAL3_BAUD,38
SERIAL3_PROTOCOL,5
SERIAL4_BAUD,57
SERIAL4_PROTOCOL,4
SERIAL5_BAUD,921600
SERIAL5_PROTOCOL,1

magicrub, 2016-08-29:
TomL in case you didn't know, if you change any of the SERIAL_ params
you'll have to reboot for them to take effect. Just making sure the params
Luis sent you were in effect.

lvale, 2016-08-29:
Thanks, Tom. I assumed that too

tomlauzon, 2016-08-29:
Just trying to isolate the problem. On our bench we have nothing else connected. If it turns out that the problem is that other equipment are causing trouble we'll try to figure out why that is. I'm still not sure why it's working here and not on your side. This other guy had a similar problem with another pull request for FrSky #3786, not related to option 10. Could be the same issue...
Could you please try disconnecting all equipment, not just setting params at 1? I don't mean to be a pain...
Thanks

Thanks @magicrub, we tried that as well ;)

lvale, 2016-08-29:
Let's be clear - The only equipment connected to the PixRacer is the absolute minimum to use the controller - ie GPS, Power module, Safety Switch, Buzzer, ESC's, SBus receiver and the FrSky cable for telemetry. Which do you suggest to remove?

tomlauzon, 2016-08-29:
I would say try to remove one at a time until it works. I would start with the GPS. If you've removed everything, then the problem is somewhere else and is harder to troubleshoot because we will have the exact same configuration as you: Pixracer + cable + Taranis. Then the problem is either with the firmware, with the Pixracer, with the receiver or with the FrSky cable. We need to isolate the source of the problem.

lvale, 2016-08-30:
Sorry..... It's completely out of line as a troubleshooting procedure, because everything else works just fine, even the other telemetry options.

I'll just wait for the next users...

tridge, 2016-08-30:
@tomlauzon can you please explain how instance numbers are allocated in the protocol? I don't see any instance IDs in the code, so what is the mechanism in the protocol for multiple instances showing up for a ID of 0x5001 ?

magicrub, 2016-08-30:
It's completely out of line as a troubleshooting procedure

Why? Sounds pretty normal to me.. You guys should probably take this offline, perhaps chat in real-time on mumble

lvale, 2016-08-30:
@magicrub Removing the minimum components required to operation, even when it works perfectly with all the other options, from my point of view is not a reasonable troubleshooting. But I might be wrong, as my wife frequently tells me. I guess that removing the power module might be a good start Tom :)

magicrub, 2016-08-30:
@lvale removing things one at a time until it works, then adding them back one at a time, is a fundamental way to troubleshoot something. I have to agree with your wife on this one :)

magicrub, 2016-08-30:
This was closed by @lvale , is it working? Did you find the problem you were having?

lvale, 2016-08-30:
Not working.

magicrub, 2016-08-30:
then why is this issue closed? Mistake?

magicrub, 2016-08-30:
This problem is still outstanding

lvale, 2016-11-06:
@tomlauzon 
Any news on the wiki page ?

tridge, 2016-11-06:
@tomlauzon and @floaledm, any progress on this?

floaledm, 2016-11-07:
@tridge Thanks for checking. We'd like to know what is causing this as well... We would require lvale to perform troubleshooting as we are not able to replicate on our PixRacer.

That being said, this issue does not occur in OpenTX 2.2 (currently RC7) since sensors in the 0x50xx range are not discoverable (the sensor data in that range is directly fed to the lua scripts).

OXINARF, 2016-11-07:
> That being said, this issue does not occur in OpenTX 2.2 (currently RC7) since sensors in the 0x50xx range are not discoverable (the sensor data in that range is directly fed to the lua scripts).

Well most probably that just hides a bug instead of fixing it.

> We would require lvale to perform troubleshooting as we are not able to replicate on our PixRacer.

Luis probably can do some tests but you need to say what you need him to do.

floaledm, 2016-11-07:
We did: https://github.com/ArduPilot/ardupilot/issues/4744#issuecomment-243282419

magicrub, 2016-11-07:
> We did: #4744 (comment)

Yup, he did.

floaledm, 2016-11-07:
@lvale Could you test whether this still occurs as of the current master? If it does, could you please at least go through the suggested troubleshooting steps https://github.com/ArduPilot/ardupilot/issues/4744#issuecomment-243282419 so that we can identify the origin?

lvale, 2016-11-07:
ok, I bit the bullet. Again. Using master on PixRacer with X4RSB receiver with telemetry connected on the PixRacer at serial4. Only peripheral connected besides the receiver is the Power Module, ie no GPS and no external mag. Motor outputs also disconnected from the PixRacer. I believe this is the absolute minimum required to operate, but it isn't flyable.

Serial port 4 option set to 4 (FrSky telemetry SPort) the radio correctly detects all the sensors, although both the discovery process and the refresh of data is slower than on 3.3.3

Changed the serial 4 option to 10, reboot the PixRacer and after erasing the previous sensors on the telemetry setup screen, started the sensor discovery process. After about 20 minutes of sensor discovery all the 32 slots get full. So slower than last test. Lots of repetitions of sensor ID's on the 5000 range (the custom range). 25 minute video anyone ?

floaledm, 2016-11-08:
Thanks lvale for testing and reporting back! I didn't know you ran discovery for that long (I typically don't run discovery for more than 5-10 seconds), and that's probably why I wasn't seeing the problem, so now there's a chance I could replicate it and ultimately find the source...
Using your "absolute minimum" setup, that amounts to approximately one packet per minute being sent on the wrong instance (a.k.a. sensorID) which is ~0.05% of packets but definitely not 0%. I suppose there's room for improvement even though there is no need for discovery and the instance is ignored for packets in the 0x50xx range in OpenTX 2.2, but more importantly I think identifying the source could help assess whether this could cause actual problems down the road.

closed 2018-08-19 by lvale