ArduPilot/ardupilot #4745: EKF pred_horiz_pos_abs bit set to 0

expntly, 2016-08-29:
#### Issue details
The EKF_STATUS_REPORT message I receive has the pred_horiz_pos_abs bit set to 0.  I'm trying to figure out why, this is preventing me from arming properly (from dronekit).

Pixhawk's light is flashing green.  Can be turned solid green by pushing safety button and full right yaw, which arms.  Switches back to disarmed (flashing green) a dozen seconds later.

This seems related: https://github.com/ArduPilot/ardupilot/blob/d1ecc63bf154d41dcbdf97f412b9a9ce6e429728/libraries/AP_NavEKF2/AP_NavEKF2_Control.cpp#L422

My GCS shows

>   GPS Count: 6
>   GPS Lock: 3D DGPS Lock
>   HDOP: 3.0
>   VDOP: --.--

#### Version
ArduCopter 3.3.3

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
quad

#### Hardware type
pixhawk

expntly, 2016-08-29:
I traced it back to poor GPS performance, as calculated in NavEKF2_core::calcGpsGoodToAlign().

> GPS horiz error 10.1m (needs 5.0)
> GPS HDOP 3.4 (needs 2.5)

I'm doing tests indoor so I might be asking for trouble.  Is this the expected behavior?

Also, any idea where in QGC or MP prearm_failure_reason can be seen?  I had to add code into the firmware to print it manually and flash.

OXINARF, 2016-08-30:
This is a support question and those should be posted at the forum (http://discuss.ardupilot.org).

Anyway, the pre-arm failures should be presented in the GCS. I haven't used QGC but I believe there's a place where messages are shown. MP shows it in the HUD and also has a messages tab where you can see all messages ArduPilot has sent.

closed 2016-08-30 by OXINARF