ArduPilot/ardupilot #4752: OSX: build failing due to missing byteswap.h

gmorph, 2016-08-30:
I'm not the OSX expert and am just helping someone but OSX doesn't have byteswap.h as far as I can tell.  Its this line that's killing the build
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/utility/sparse-endian.h#L23
and probably the following endian.h as well as I don't think OSX has that either.
This change to that file occured on 9/7/2016
-#if !defined(HAVE_BYTESWAP_H) || HAVE_BYTESWAP_H
 #include <byteswap.h>
-#endif

Can we put that back in as it means the OSX compile works.

Possible other solutions could be:
https://github.com/jgarzik/pushpool/pull/33/commits/d8d8e37763dbf3f7a0f53e340845c494d46668c5
http://debugjournal.tumblr.com/post/99661010362/missing-byteswaph-and-endianh-on-mac-os-x

Thanks, Grant.

lucasdemarchi, 2016-08-30:
Please provide compiler and its version. For gcc minimum supported version is 4.9.... Clang I think is 3.6, but only for CI and sitl

gmorph, 2016-09-02:
Lots of discussions have happened that we haven't pasted here :-)
This only happens when I try to build SITL.  px4-v2 works fine.
I'm using gcc/g++ on OSX and build with these commands:
./modules/waf/waf-light configure --board sitl --check-cxx-compiler g++ --check-c-compiler gcc
./modules/waf/waf-light build --target bin/ardurover

gmorph, 2016-09-02:
Output from configure step is:
Setting top to                           : /Users/grant/uav/ardupilot 
Setting out to                           : /Users/grant/uav/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Checking for 'g++' (C++ compiler)        : /usr/local/bin/g++ 
Checking for 'gcc' (C compiler)          : /usr/local/bin/gcc 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for program 'python'                  : /Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python 
Checking for python version >= 2.7.0           : 2.7.11 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/local/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
'configure' finished successfully (1.909s)

lucasdemarchi, 2016-09-02:
> Checking for header byteswap.h : yes

This is wrong. The configure should tell you don't have that header.... See https://github.com/ArduPilot/ardupilot/pull/4773#issuecomment-244481504

lucasdemarchi, 2016-09-03:
Should be fixed in #4776. If you still experience issues, please reopen. Thanks.

closed 2016-09-03 by lucasdemarchi