ArduPilot/ardupilot #4759: Waypoints saved to SD card to increase count from 724

freyapilot2002, 2016-09-01:
#### Issue details
Is it possible to save the waypoints to the SD card rather than the eeprom?
We have many missions that using do_digicam control is not a viable option.
Previous request was closed because of the increase to 724 points. This is not enough. We have several areas that require close to 4000 waypoints because of our 9m camera triggering.

#### Version
3.2.1 Octa Copter X8

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
X8 Octa Copter

#### Hardware type
PixHawk

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2016-09-01:
> 4000 waypoints because of our 9m camera triggering
Do you know about the trigger interval? One trigger to auto-toggle and another to turn it off.

closing, duplicate of https://github.com/ArduPilot/ardupilot/issues/4424

closed 2016-09-01 by magicrub

WickedShell, 2016-09-01:
Why is the interval trigger not sufficient? Also worth noting that transfer
times will be extremely high on 4k waypoints.

freyapilot2002, 2016-09-01:
Hello,

 

Because of the cost of the vaccine pellets we are dropping, we need to have total control over where the pellets are dropped.

We have GIS shape files and there is a lot of areas with travel time between the points where we need pellets dropped. We tried using distance trigger but we were dropping pellets way too often and in places they were not needed.

We are ok with a long transfer time because that will give us time to swap batteries and reload pellets.

We have about 26 minutes of flight time with payload and we are only using about 12-14 with the 724 limit.

Next stage is to use a fixed wing and we will need the higher counts while we stay in the air for an hour.

 

Would you be willing to help write code to do this?

 

Thank you,

Kurt

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2016.0.7752 / Virus Database: 4649/12920 - Release Date: 08/31/16

gmorph, 2016-09-01:
Sounds like an interesting project - do you have a link to a website?

Thanks, Grant.

freyapilot2002, 2016-09-01:
Hi Grant,

 

We are working with local Fish and Wildlife agencies to save the Black Footed Ferret.

 

Here’s a link to some stories about it.

It went a bit crazy because some reporter misquoted that we are using M&M’s for the bait. Not true.

 

http://www.wired.com/2016/07/fleet-mm-shooting-drones-black-footed-ferrets-last-hope/

 

https://www.theguardian.com/environment/2016/jul/12/us-government-black-footed-ferret-mandm-vaccines

 

We are going to use r/c helicopters or fixed wing for the next phase. I have a new dispenser that shoots 3 at a time. One left 9 meters, one right 9 meters and one straight down.

Will need the ability to load a lot of waypoints and digicam control points with one hour or more flight times.

 

I am looking also to have MAVlink feedback to the mission planner from our dispenser.

We have it set up so an Ultra bright LED flashes each time a pellet is dropped. Only triggers if pellet breaks the beam.

Gives us visual verification nothing is jammed. The pellets are not as round as we specified and it has resulted in quite a challenge on the dispenser development.

Would be great to send a tone or a light blink on the mission planner as a secondary notification. Tone would be best so we can keep eyes on the craft.

Gets tiresome looking through the binoculars when the copter is on mile long transects. The area we are treating is really flat and you can still see the copter(it’s just a dot) but not the light unaided.

 

Kurt

iskess, 2016-09-01:
Kurt,

I submitted a Mission Planner GitHub request to display each time a photo was triggered. Michael Oborne kindly added this feature recently. If your pellet dispenser it triggered by the cam trigger, you will get a green camera icon each time it attempts to drop a pellet. This won't tell you if the pellet is jammed however. You can use CAM_FEEDBACK_PIN and your bright LED trigger sensor to record each time the pellet actually dropped in the dataflash logs. 

Iam

freyapilot2002, 2016-09-01:
Hello Iam,

 

Thank you for the quick response.

Do you feel it would be possible to use the CAM_FEEDBACK_PIN to make an audible through MAVlink?

And what are your thoughts on the increased waypoint function?

 

Best regards,

Kurt

 

Version: 2016.0.7752 / Virus Database: 4649/12909 - Release Date: 08/30/16

iskess, 2016-09-01:
It's certainly possible, but mavlink has latency and dropped packets. If you watch during a mapping mission, you will notice s few of the green camera icons get missed. Looking back at the dataflash logs you see that the camera did fire, but Mission Planner didn't catch it. 
I use FPV on all my aircraft. I use the audio channel to hear my onboard voltage alarm which I trust more than MP, especially because it tells me about every cell. You could set it up to use the audio channel to send feedback from your "glorified gum ball machine". This would be more reliable.  

I'd love to see more waypoints. There are case I've run into where CAM_TRIGG_DIST is not desired. Mapping in very mountainous terrain sometimes you can't use the automatic Survey Grid and you need each photo location to be exactly on point.  
In years past the solution discussed was to allow the SDcard to hold multiple flight plans, and auto load new flight plans at a given point. This would also help with the goal of having copter fly numerous missions autonomously, complete with precision landings and robotic battery changes. 

Your project sound very fun and satisfying!  

Iam

freyapilot2002, 2016-09-01:
That's not a bad idea. I could use a little arduino and have it compare the trigger and the LED output. If LED does not equal digicam control for X amount, sound buzzer and we could here it over a little speaker back at base.
Alright, now lets get someone to write the waypoints to the SD card and we are in business. :-)

magicrub, 2016-09-01:
> black-footed-ferrets-last-hope/
> government-black-footed-ferret-mandm-vaccines
> auth/AEOY-3tA48VFrkU5ZXohuOvShljHTb0Kks5qlj5ugaJpZM4JyMBh> . <
> auth/AEOY-10ZRtsUXkEj6w90qWwBQYO0hhDoks5qlk13gaJpZM4JyMBh> . <

The way point count max is much much higher on Linux boards, May be best to
use that.

lvale, 2016-09-05:
@freyapilot2002  The PR to increase this for the Linux boards was this: 

https://github.com/ArduPilot/ardupilot/pull/4082

On Linux based boards this is, for now, 200 rally points, 1000 fence points, 7000 waypoints.

I believe the new PR is now https://github.com/ArduPilot/ardupilot/pull/4172

freyapilot2002, 2016-09-09:
Are the Linux boards as Plug and Play as the Pixhawk?

 

#4082 <https://github.com/ArduPilot/ardupilot/pull/4082> 

Version: 2016.0.7752 / Virus Database: 4649/12949 - Release Date: 09/05/16