ArduPilot/ardupilot #4760: Feature request - support for copteroplane

night-ghost, 2016-09-01:
Quad-plane is a great step but this construction is not the only possible. Switching of motors or turning of motor rack are needed for manned aircraft, but unmanned can fly "on the side". See videos
https://www.youtube.com/watch?v=FEtudz2g3AM
https://www.youtube.com/watch?v=aILcj21w3-4

This solution is more efficient because it does not need an extra motor and controller, and on takeoff is not necessary to overcome the resistance of the air from the wing.

The transition from copter mode to plane only need to consider the controller 90-degree rotation. 
Also control such a device can be without any control surfaces, only by motors trust, although the presence of control surfaces enhances the reliability.

iskess, 2016-09-01:
I'm guessing it will actually be less efficient. The props won't be optimized for both forward and vertical flight. Using 4 motors with flat pitched props will burn more energy due to high drag of the prop disks. With QuadPlane, I believe the extra motor would prove worth it's weight by being optimized for high airspeed. Meanwhile the other 4 propellers are parallel to the airflow minimizing drag when compared to 4 flat disks. Also, a "tail-sitter" design is not well suited for windy conditions. A QuadPlane with Q_VFWD_GAIN and weathervaning actually performs better with wind.  
All that said, more build options are always better.

rmackay9, 2016-09-02:
I think this request is to support something like the XCraft. http://xcraft.io/x-plusone-drone/
My understanding is the XCraft actually uses ArduPilot (Copter to be specific) but one major issue is the lack of fins to provide good yaw control (when nose is pointed up, roll control when flying with the nose horizontally).  The motors by themselves can't really provide enough rotational force to give good control.  Even with multicopters you can see the vast difference in yaw strength vs roll-pitch strength.  This is the major issue with these frame types.  If they have fins it works much better but in this case, you've then basically got a single copter (which ardupilot also supports).

cudnn, 2016-09-02:
hi. rmackay9 .i want use quadplane control xpluse one.  use ahrs.rotate to make bodyframe rotate 90.and use quadplane tilrotor to compensate  motor outputs.

night-ghost, 2016-09-02:
>The props won't be optimized for both forward and vertical flight.

yes but this is not required - we can put up with some loss of efficiency during takeoff and landing. Harrier so generally ineffective in the hover, so what?

>something like the XCraft

It seems but not completely. XCraft is more copter - but with the wing, and therefore has low flying time. But I say about the same plane substantially pure, with high lime and range, but with the ability to hover - controlled and stabilized hover. And yes, singlecopter probably more suitable scheme, but my 1st try was a small EPO wing plus quad with Baseflight and Acro mode.

>one major issue is the lack of fins to provide good yaw control

Ailerons? ;) yaw turns to roll in plane mode

cudnn, 2016-09-04:
night-ghost. .did you have test it .we can make it tegether

cudnn, 2016-09-04:
tridge is developing quadplane,  he can help us.because it is so easy,only need rotate ahrs

cudnn, 2016-09-04:
tridge is developing quadplane, he can help us

night-ghost, 2016-10-18:
>.did you have test it 

Very smal testing - stuck copter into to the flying wing by adhesive tape and a some fly in acro :) To hard for me to fly without stabilization

auturgy, 2017-11-28:
https://discuss.ardupilot.org/t/vtol-delivery-wing/22133

closed 2017-11-28 by night-ghost