ArduPilot/ardupilot #4772: Using 2 servos for pitch

AndreyKortosov, 2016-09-02:
#### Issue details
Hello, I want to use 2 servos for pitch in APM 2.6, they should spin in opposite direction. How I can change spin direction of 1 servo, since they are both connected to 1 RC output?
The possible solutions are: to add one more RC output which will be connected to 1 axis (either pitch or yaw), or make it possible to control 2 servos by 1 RC output.
I want to use 2 servos on pitch axis in order to have higher effort for better stability of antenna.
#### Version
V1.0.0
#### Platform
[  ] AntennaTracker
#### Airframe type
Fixed wing
#### Hardware type
APM 2.6
#### Logs

gmorph, 2016-09-02:
This is not a bug in the code so should really be posted on the support forums - http://discuss.ardupilot.org/.  
I assume this is for plane.
You can't do it if both servo's are connected to the same output.  You need to connect one of the servo's to a different RC output (i.e. RC6 for instance) and then set RC6_FUNCTION to 19 (the Elevator controls the Pitch) so that servo will behave like another elevator.  It moves in the wrong direction simply set RC6_REV to -1.

Thanks, Grant.

magicrub, 2016-09-02:
Also, APM 2.6 is depreciated and no new changes are going in.

AndreyKortosov, 2016-09-05:
Thanks for the answer.
Do I understand correctly that you suggest assign another RC output to the
firmware?
We also think to do, but there is no in the firmware. So we wrote on the
form and github.
Also we use APM 2.6 because there is no other, and do not want to spend on
other things.

magicrub, 2016-09-05:
Grant, it looks like this is for antenna tracker. Dual servos for double
torque but opposite pwm values because they are mounted opposite.

So, this is a new feature and would require a firmware change. Sadly it's
been too hard to maintain apm 2.x so I'm not sure how that would get
implanted for you. We can at least get this implemented for Pixhawk and you
could try and port it yourself to apm.

magicrub, 2016-09-05:
@hiro2233 had talked about possibly maintaining the APM 2.x branch, maybe you could work with him on this?

rmackay9, 2016-09-05:
I've heard this request for Tracker to support multiple servos for a single axis from others as well.  It's a good idea I think.  I might eventually get to that for the Pixhawk and other boards but I don't think I could justify personally go back to the APM boards.

hiro2233, 2016-09-08:
Hi @AndreyKortosov, ok, you don't need an extra channel if you are finding to get more torque with two servos. You could attach two or more servos that you want with an "Electrical trick" on the same channel, Antenna Tracker have already 2 channels for you and you can invert the direction if you want in AntennaTracker settngs in MissionPlanner.

How i do it the trick and why it work?
If you are using micro servos like 9g it's recommendable (not by me) use only one on each channel an MAX 2 servos directly on the board, but sincerely and personally, never in my life i connected it directly, knowing that the board share the same line power, but it's better that you source the power externally to the servos. Servos have more power consumption when it reach the top level angle and it cause interferences on the little dimension board, and maybe it could reboot the micro, for that reason you need to power externally. Standard servos can work from 4.5v - 5.8v around 200mah - 500mah.
To do the "trick" are simple you need to connect the SIGNAL wire (PWM) and GROUND from the servo to one channel RC Output1 on the board only and the line POWER and shared GROUND to the external source power, to add 1 more servo, you need to do the same steps with this second servo. The same way you can do for the RC Output2.

It's all.
Good Luck!

AndreyKortosov, 2016-09-15:
Hello @hiro2233, thank's for the answer, but you do not understand me.
We have 3 servo on 2 axis: 1 servo - yaw, other 2 servo - pitch for more torque. And we need these servos for pitch reverse, because they are located at different edges of the platform, and one direction of movement.
Servos are powered by external power supply.

Naterater, 2019-01-08:
It would be nice to see the SERVOn_ features on all vehicles to prevent this issue in the future.

IamPete1, 2019-02-05:
tracker on master does have SERVOn_ so the problem would be fixed in master

IamPete1, 2019-05-06:
new beta fixes this

closed 2019-05-06 by IamPete1