ArduPilot/ardupilot #4785: Fatal error: 'atomic' file not found during building

julien10, 2016-09-05:
#### Issue details
I'm trying to build ArduCopter on a Snapdragon Flight, but I'm getting the following error:

> fatal error: 'atomic' file not found

#### Version
3.4

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
/

#### Hardware type
Snapdragon Flight

#### Logs
```
%% libraries/AP_HAL_Linux/qflight/dsp_functions.do
%% libraries/AP_HAL/utility/RingBuffer.do
// BUILDROOT=/tmp/ArduCopter.build HAL_BOARD=HAL_BOARD_LINUX HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_LINUX_QFLIGHT TOOLCHAIN=QFLIGHT EXTRAFLAGS=-DGIT_VERSION="6e2c600f" -I/home/ja/Snapdragon/ardupilot/libraries/AP_Common/missing -DMAVLINK_PROTOCOL_VERSION=2 -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE -DHAVE_ENDIAN_H -DHAVE_BYTESWAP_H -DHAVE_CMATH_ISFINITE -DNEED_CMATH_ISFINITE_STD_NAMESPACE -D__STDC_FORMAT_MACROS
%% control_brake.o
In file included from /home/ja/Snapdragon/ardupilot/libraries/AP_HAL/utility/RingBuffer.cpp:4:
/home/ja/Snapdragon/ardupilot/libraries/AP_HAL/utility/RingBuffer.h:3:10: fatal error: 'atomic' file
      not found
#include <atomic>
         ^
1 error generated.
../mk/board_qflight.mk:88 : la recette pour la cible « /tmp/ArduCopter.build/libraries/AP_HAL/utility/RingBuffer.do » a échouée
make: *** [/tmp/ArduCopter.build/libraries/AP_HAL/utility/RingBuffer.do] Erreur 1
make: *** Attente des tâches non terminées....
In file included from /home/ja/Snapdragon/ardupilot/libraries/AP_HAL_Linux/qflight/dsp_functions.cpp:47:
/home/ja/Snapdragon/ardupilot/libraries/AP_HAL/utility/RingBuffer.h:3:10: fatal error: 'atomic' file
      not found
#include <atomic>
         ^
1 error generated.
```

guludo, 2016-09-06:
Unfortunately I don't have the toolchain for this board :/

OXINARF, 2016-09-06:
@guludo Which is why I assigned it to @tridge :smiley:

josephyuzb, 2017-09-01:
hello guludo,
have you solved it ?I ve met the same issue as you using the snapdragon 810 drone 
Thanks.

WickedShell, 2018-06-25:
QFlight support was removed, due to lack of maintenance (and existing bugs).

closed 2018-06-25 by WickedShell