ArduPilot/ardupilot #4789: Copter-3.4: motor twitch when disarmed when using OneShot

rmackay9, 2016-09-06:
reported by Luis including this video: https://www.youtube.com/watch?v=w5QLcQU7Xro

lvale, 2016-09-06:
@rmackay9 @tridge 

ok. more testing, this time with logic analyser attached to each of the 4 ESC's

Debugging the Oneshot125 twitching.

Used Flight controllers and ESC's
	PixRacer - ArduPilot Copter 3.4RC4
	CC3D Atom - Betaflight 2.9.0
	BLHeli Suite 16.3.14.7.0.2 
	ESC used - Blueseries 12A for Multicopter Motors - BLHeli revision 14.7 (Atmel based)
	
Test procedure

	PixRacer
		MOT_PWM_TYPE=0
		MOT_PWM_MAX=2000
		MOT_PWM_MIN=1000
		
		ESC calibrated as per wiki
		
		No problems observed
		
	PixRacer
		MOT_PWM_TYPE=1
		MOT_PWM_MAX=2000
		MOT_PWM_MIN=1000
		
		ESC calibrated as per wiki
		
		No problems observed
	
	PixRacer
		MOT_PWM_TYPE=2
		MOT_PWM_MAX=2000
		MOT_PWM_MIN=1000
		
		ESC calibrated as per wiki
		
		Motors twitch after Safety switch pressed, but still nor armed
		
	CC3D Betaflight
	
		Setting Motors  - Normal (equivalent as MOT_PWM_TYPE=0)
		
		ESC calibrated
		No issues,
	
	CC3D Betaflight
	
		Setting Motors  - OneShot125 (equivalent as MOT_PWM_TYPE=2)
		
		ESC calibrated
		Motors twitch as with ArduPilot
		
	
	Bypass connection directly with BLHeli suite
	
		ESC calibration done between 1000 and 2000 on the app
		
		Motors twitch

I would say that there might be an issue with several ESC's (both Atmel and Silabs) and Oneshot125. 

I still have to perform a few more tests, to see if I can get a workaround.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[motor test.mov.zip](https://github.com/ArduPilot/ardupilot/files/457396/motor.test.mov.zip)
[Logic traces.zip](https://github.com/ArduPilot/ardupilot/files/457397/Logic.traces.zip)

lvale, 2016-09-06:
Further tests

As suggested on ardupilot.org forum, did try with safety switch disabled, and redid the ESC cal with MOT_PWM_TYPE 0, 1 and 2, and results are the same. OK with 0,1 twitching with 2.

Did try redoing ESC cal with MOT_PWM_MAX and MIN to the BLHeli suite suggested default values 1148 and 1832.
Results slightly different: This time regular OneShot took a long time to initialize the ESC's, every time that the vehicle started and the safety switch was pressed. OneShot 125 still the same twitching...No problem with regular PWM

And another one. ESC cal with MOT_PWM_MAX and MIN to 0 to go back to sending the Throttle PWM values (my radio - Taranis 982/2002)
Results slightly different: This time regular OneShot took a long time to initialize the ESC's, every time that the vehicle started and the safety switch was pressed. OneShot 125 also took a long time to signal the ESC's ready (30+ seconds) and then still the same twitching... Same delay on initializing the ESC's with regular PWM but less delay 5/10 seconds.

Curious that not all the ESC's took the same time to be initialized, and it was a random pattern, sometimes ESC1 was the first others ESC 4 was the first, etc, etc...

So redid the initial config MOT_PWM_MIN/MAX 1000/2000 MOT_PWM_TYPE to regular Oneshot (1) redid the ESC calibration and getting everything ready to fly tomorrow with RC4... (I hope).

I'm going to ask the BLHeli people if this is expected behaviour... we'll see....

lvale, 2016-09-06:
With all this testing I also made a sugestion for Tower :) https://github.com/DroidPlanner/Tower/issues/1802

rmackay9, 2016-09-06:
Great, thanks very much for testing this.

So I think because the twitching is also found when using the CC3D Betaflight software, we can conclude that it's very likely to be either the hardware or the BLHeli software.

lvale, 2016-09-06:
Not only on Betaflight, but also using the BLHeli native tool to calibrate ESC's

lvale, 2016-09-06:
Question raised on BLHeli thread at RCG http://www.rcgroups.com/forums/showpost.php?p=35663199&postcount=9846

lvale, 2016-09-09:
I'll try with those values via MP, but I've used the BLHeli suite app and withe the suggested values it was twitching... 

be back in a bit :)

lvale, 2016-09-09:
ok, Yes it works like @tabascoz said.

The problem is that although the ESC's only change to OneShot125 mode when they boot and get initialized to the signaling they receive at boot time. So although we can change freely the PWM_OUT mode between 1,2,3 without rebooting the PixHawk, the ESC's were initialized at Normal mode.

So better testing must be: change settings to OneShot125 on ArduPilot and adjust the PWM_MIN to whatever value you need. then power off everything and power on. As we are sending OneShot 125 signal then the ESC's get initialized to that, and the twitching occurs no matter the value we send.

lvale, 2016-09-09:
Following on this theme...

So I think I found a workaround this issue.

Set your MOT_PWM_TYPE to 2 (OneShot125)

Now the values below have to be found by trial and error

Set the MOT_PWM_MIN to 1100

Calibrate the ESC's

Don't worry motors still twitching..

Power down, and power up

Now go to MOT_PWM_MIN and set it to a value below what was used for calibration, say 1000

Motors stop twitching because they are receiving a signal below their calibrated minimum.

Now arm, and check if the motors spin normally. In my case with the values above they were twitching, because the value they were receiving was very close to the minimum they were calibrated, so I played around with the MOT_PWM_MIN value used to calibrate the ESC until I was satisfied...

ok, It's a bit twisted process, and I'm not sure I made myself clear.

lokanaft, 2016-09-09:
It's not a oneshot problem.
My setup work good after press safety with px4.io firmware with oneshot125.
And ardupilot work good, if disable safety switch.

lvale, 2016-09-09:
> >It's not a oneshot problem.
> My setup work good after press safety with px4.io firmware with oneshot125.
> And ardupilot work good, if disable safety switch.

Please tell us on which values the PWM_MIN was calibrated on the ESC (assuming the calibration was performed also using OneShot125, and which values are on the output of the Flight controller with either PX4 and ArduPilot before and after the safety switch is pressed, and also when the safety switch is disabled

My testing without any flight controller plugged in to the ESC - direct from a USB/Com interface and only using the BLHeli config tool the twitching was repeatable.

lokanaft, 2016-09-10:
My esc config:

[image]

I calibrate it as in MissionPlanner instruction

lvale, 2016-09-10:
@lokanaft 
Thanks. On a first glance the main differences, that could be related to this, are only the BLHeli version (14.6 vs 14.7), the PWM Frequency/Damped which is the active motor braking (Damped light vs Low), the Motor Timing (Medium Low vs Low) and perhaps also the Motor Startup Power (0,5 x 1.0).

Except for the version of BLHeli, I'll test the other differences and report back

rmackay9, 2016-10-13:
@lvale, it's ok to close this right?  Because I think we've determined that it's not an issue with ardupilot.
I'll close but we can re-open if I've got that wrong.

closed 2016-10-13 by rmackay9