ArduPilot/ardupilot #4796: Plane: Handle failsafe differently for AUTO than other modes.

estebanavv, 2016-09-07:
#### Issue details
_Please describe the problem, or desired feature_

Currently there is only two global parameters that control the failsafe behavior regardless of the flying mode.
FS_SHORT_ACTN
FS_LONG_ACTN

For users that want to have long missions and/or fly in other modes just for the pleasure of flying in the same session it creates a problem. On long missions I would like failsafe to continue the mission but if I am in FBWA and signal is loss, or there is a TX failure, the plane will continue on with little or no possibility to regain control. 

What I am suggesting is to create two new parameter that will only apply when the failsafe occurs in AUTO mode.

FS_AUTO_SHORT_ACTN 
FS_AUTO_LONG_ACTN

If these parameters have no values then it could take the values of the first two.

#### Version
_What version was the issue encountered with_
All
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Any fixed wing

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Any

#### Logs
_Please provide a link to any relevant logs that show the issue_

iskess, 2016-09-07:
It already does exactly what you want. Set it to continue the mission and it will still RTL if not in Auto.

estebanavv, 2016-09-07:
Oops sorry for opening the ticket. Thank you.

closed 2016-09-07 by estebanavv

magicrub, 2016-09-07:
Makes sense, thanks for the idea.