ArduPilot/ardupilot #4798: Copter 3.4 - Aux Channels inconsistent behavior with Safety Switch

lvale, 2016-09-08:
#### Issue details
Aux channels don't work when safety switch is off.
After pressing safety switch the AUX channel becomes active 
Pressing safety switch again does not "switch off" AUX channels.

Inconsistent behavior

#### Version
3.4RC4
#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad
#### Hardware type
PixHawk
#### Logs

rmackay9, 2016-09-08:
As discussed, there are two issues perhaps:
1. a bug in which the safety switch isn't stopping the output to the auxiliary channels
2. we should change the default in copter so that for the regular pixhawk board, the auxiliary channels (normally used for gimbals, landing gear, etc) are enabled by default.

netptl39, 2016-09-13:
Additionally on AUX channels the if RCx_FUNCTION=0 and 1 the RCx_ TRIM is disregarded so output PWM is 0 even after Switch is ON

lvale, 2016-10-31:
To be tested with 3.4 release

rmackay9, 2016-11-19:
I think this is fixed.  @lvale, no pressure but could you give it a try with AC3.4.2?

OXINARF, 2017-06-04:
@lvale Is this still valid?

lvale, 2017-06-04:
Issue solved with channel mask PR

Closing

closed 2017-06-04 by lvale