ArduPilot/ardupilot #4799: Copter - Confirmation beep on parameter change

lvale, 2016-09-08:
#### Issue details
Vehicle should acknowledge with a beep when parameter is changed
#### Version
New feature
#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A
#### Hardware type
PixHawk
#### Logs
N/A

DonLakeFlyer, 2016-09-09:
Curious as to why this would be needed. Going to lead to lots of beeping that may not make sense to anyone.

lvale, 2016-09-09:
@DonLakeFlyer this happens on PX4 :)

squilter, 2016-09-09:
I agree with @lvale. This is default behavior with PX4 and it is very convenient.

squilter, 2016-09-11:
Actually it looks like PX4 just removed this feature. See PX4/Firmware#5331

lvale, 2016-09-12:
Since we already have a parameter to silence the beeping, @tridge suggestion to have a extra value to it is IMHO the best solution - and yes option 4 would play mp3 files stored on the sdcard using the buzzer :)

DonLakeFlyer, 2016-09-13:
> There's a GCS operator and a RC operator and for every change done during tuning, the RC operator acknowledges when the vehicle beeps for a parameter change, so that he is expecting a different behavior.

Keep in mind that QGC itself will tell you if a parameter write fails.

lvale, 2016-09-13:
@DonLakeFlyer Usually the GCS operator and the RC operator are at a distance that might prevent the communication.

closed 2019-01-27 by lvale