ArduPilot/ardupilot #4803: Copter: lightware rangefinders I2C connection is blocking on Pixhawk

rmackay9, 2016-09-08:
Although we support the lightware range finder's I2C connection, we don't recommend that it be used on Pixhawk flight controllers especially if using the Copter code because the I2C calls are blocking and are called from the main vehicle code.  Linux boards do not have this problem.

rmackay9, 2016-09-08:
I've tentatively assigned this to Lucas but I can un-assign if that's incorrect.

lucasdemarchi, 2016-09-08:
PX4 is missing the implementation of these 2 APIs:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL/I2CDevice.h#L62-L68

I can help on reviewing, but I don't have plans to implement it soon.

rmackay9, 2016-09-09:
Ok, Lucas thanks for that.  I've removed you as the assignee.  It felt weird that you would do the px4 implementation from the start actually.

rmackay9, 2016-11-22:
This should now be solved in master, we just need to test.

silkstone, 2017-02-09:
Hello, I was just wondering what recent results you have had when testing the lightware rangefinders through the i2c with the updated master?

OXINARF, 2017-02-09:
We have good reports from people testing 3.5-rc1.

rmackay9, 2017-02-15:
Yes, we can close this!

closed 2017-02-15 by rmackay9