ArduPilot/ardupilot #4811: Copter: Landing gear should not retract when vehicle is on the ground

jason406, 2016-09-10:
#### Issue details
Landing gear should not retract when vehicle is on the ground (disarmed or armed), in case of misopration. If the landing gear control switch is accidentally switched to retract position when vehicle is still on the ground, the airframe or gimbal will be damaged. 

#### Version
Arducopter 3.3.3
#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Hardware type
Pixhawk

rmackay9, 2016-09-10:
That's an interesting idea.  It's not technically hard I think to make sure that it's always deployed when landed.  My only concern would be that it could make it a little harder to test the feature and the gear would deploy immediately upon startup in cases where it's not already deployed (i.e. in case it was stored with landing gear retracted in order to fit into the back of a vehicle).  Part of the problem is that the flight controller doesn't know the state of the gear because we have no feedback from the servos...

jason406, 2016-09-12:
I just figured out a technical solution to the conflict between misopration and storage. In general case, the landing gear is always deployed when landed. If the operator does want the landing gear retracted,  he or she may toggle the switch on RC several times continuously. Because misopration happens only once.

sgofferj, 2016-12-23:
The hardware switch for landing detection I proposed in #523 would make this a bit easier :)

Pedals2Paddles, 2020-02-06:
There is now weight on wheels switch handling, mavlink control of landing gear, 3-position switch control of landing gear, and automated landing gear operation based on altitude and flight mode. So I believe this one is covered now :)

closed 2020-02-06 by Pedals2Paddles