ArduPilot/ardupilot #4812: FrSky: add real-time tuning options

magicrub, 2016-09-11:
This is mainly for plane but can be any vehicle.
For tuning an aircraft, it's nice to tweak a param then see the behavior (and repeat). It's also very handy to see the behavior in data form such as a plot in real-time. Right now that is done with a GCS but it would be great if it was added to FrSky telem. 

Wish List:

- [ ] plot various datas on the screen: height demand and also actual height (and rpy throttle and a zillion others)
- [ ] be able to adjust and transmit a param from FrSky to ArduPilot
- [ ] add UI for a user to select a param to change. Presumably it would request all params, or only certain tuning ones, and display them in a list where the user can then select them.

Copter has some of these ability hard-coded a;ready, would be nice to make them more flexible in the FrSky and to plot. This idea came from the great folks at CraftAndTheory during a discussion at the InterDrone 2016.

OXINARF, 2016-09-11:
> be able to adjust and transmit a param from FrSky to ArduPilot

> add UI for a user to select a param to change. Presumably it would request all params, or only certain tuning ones, and display them in a list where the user can then select them.

This looks interesting but how would it work? Isn't FrSky telemetry one way only?

magicrub, 2016-09-11:
They have some 2-way ones, or will soon.

OXINARF, 2016-09-11:
Interesting, I couldn't find anything about it, but I look forward to it.

lvale, 2016-09-12:
Most of this "wish list" could be done IF the FrSky would send MAVLink messages. With some clever programming on the Radio side (mostly done in LUA)and sent to the vehicle, the FrSky lib would then be responsible to decode those MAVlink messages and "insert" them on the general code. I'm not sure that even the latest FrSky radio (Horus) has enough programming capacity to implement a comprehensive solution. The current radios X9D and X9E are already at almost the limit just to process telemetry data from the vehicle, so implementing a "quasi" GCS on the radio might prove a challenge.

The way I see it (at this point) is for the OpenTX/FrSky radios to send Mavlink on the telemetry link (uplink). From there it would be similar to any other connection (like regular telemetry radios)

magicrub, 2016-09-12:
I would love that to be the case. It would also help future proof it.

lvale, 2016-09-12:
There's an excellent work already done using either a Teensy or Arduino to encode and decode the FrSky telemetry data here http://www.rcgroups.com/forums/showthread.php?t=2245978

OXINARF, 2016-09-12:
Sending MAVLink through the FrSky link doesn't make any sense, it would be terrible slow. That is also said in the issue you linked to.
Also, to do most of this list it needs an uplink in the FrSky hardware, separate from the RC link, which, as I said before, doesn't exist in any of the current hardware I'm aware of.

lvale, 2016-09-12:
@OXINARF I restrained myself from commenting on the proposed solution. :)
Also the speed of the link (depending of what would be used) I'm sure wouldn't be a problem for adjusting simple parameters or settings...
But, again, it's mostly a work to be done on the Taranis side of the problem.

OXINARF, 2016-09-12:
So apparently SPort is bidirectional but not implemented yet: http://openrcforums.com/forum/viewtopic.php?f=45&t=8625&p=115856#p115856

xros, 2016-09-14:
@OXINARF You can think that lots of guys are using OpenLRS transmitters/receivers which have telemetry enabled. And the baud rate of a OpenLRS transmitter can  be set at 4800, 9600, 19200. The mavlink protocol uses baud rate at 9600, 57600, 115200 [doc](http://qgroundcontrol.org/mavlink/start). Lots of people want native mavlink protocol.  see: https://github.com/opentx/opentx/issues/2354

I know that The baud rate of FrSky telemetry is very slow at 1200. Honestly I don't really care this terrible low bandwidth telemetry on FrSky.

Please, we don't have to think only in a Fr-Sky product box. We can build transmitters/boards/system anytime. Many hardware hackers/experts/engineers need this functionality for testing and controlling drones remotely.  Years later, we will have/build/buy other radios. OpenTX is the key which lasts forever rather than a unique brand radio product...

lvale, 2016-09-14:
I think this should be closed until OpenTX can send Mavlink messages on the FrSky telemetry link.

OXINARF, 2016-09-15:
@xros Yeah, this issue is about using FrSky. I'm not against MAVLink in OpenTX, I'm against using it over a FrSky link.

@lvale I disagree, we don't need MAVLink messages in the telemetry link, we need the uplink to be done in OpenTX and then we define a protocol (id X -> does X, value in that ID is Y).

Naterater, 2019-01-18:
https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443

As this exists, I think piggybacking on that might be the best chance of getting this done.

yaapu, 2019-09-06:
PR https://github.com/ArduPilot/ardupilot/issues/4812 is my attempt at this

peterbarker, 2021-07-30:
This was implemented by @yaapu

closed 2021-07-30 by peterbarker