ArduPilot/ardupilot #4819: UAVCAN new device

RoverHunter, 2016-09-12:
#### Issue details
Is there any tutorial on how to add a CANBUS device to the project? Where we can choose the data struncture in and out to the new CAN device?

#### Version
38d51ddb322ea500b5d850b7274d61d0261f2909

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

magicrub, 2016-09-12:
Not that I know of, but looking at existing drivers is a good start. This is a better issue for the wiki issue tracker rather than the code base issue tracker. Please start a new issue there and we'll see about documenting it better. closing this issue..

closed 2016-09-12 by magicrub

RoverHunter, 2016-09-15:
I am sorry magicrub (I am new here), how can I start a new wiki issue?

gmorph, 2016-09-16:
Probably best to jump onto the gitter channel and ask there.  It would be great if you have a branch where you have been committing your work so we can see the code and help.