ArduPilot/ardupilot #4820: Log files with time issues

lvale, 2016-09-12:
#### Issue details
Log files might present out of order lines

#### Version
3.4RC4

#### Platform
[  ] All
[  ] AntennaTracker
[ x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad
#### Hardware type
PixRacer/PixHawk
#### Logs

peterbarker, 2016-09-13:
I've found the cause of this problem.

I've no idea why this hasn't been a problem for a very long time.

rmackay9, 2016-09-13:
This is in master now so closing.
This will go out with -rc5.

lvale, 2016-09-19:
The gremlin is back :)
any of the lof files here https://drive.google.com/open?id=0BzRxu4IwHPYCR3k4Mkg4WUtRNzg

peterbarker, 2016-09-21:
No, different gremlin.

Anything that goes out to the log between https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2.cpp#L571 and https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2.cpp#L630 will have a timestamp ahead of the messages written out in check_log_write.

You can see several `send_statustext_all` entries between the two lines mentioned (which will result in a `Log_Write_MSG`), but the ones actually coming out are in `InitialiseFilterBootstrap`.

One option we have here is to clamp all message times to `imuSampleTime_us` if it is non-zero.  That might be rather a large change, 'though, and each message involving a timestamp will essentially need to do a `this->now()` call rather than calling` AP_HAL::micros64`

OXINARF, 2016-09-28:
@peterbarker Isn't the issue here that we have different threads writing to the dataflash log? And we want some messages (coming from the EKF) to be written with the same time.
I think the correct way would be to have DataFlash write the time for every packet instead of having each packet have their own definition of time. Then we could have a method to freeze and unfreeze time.

peterbarker, 2016-10-05:
@OXINARF I sincerely hope we don't have multiple threads attempting to write dataflash messages; there will be all sorts of fun races in there if we do.

Having dataflash write out the time is an interesting concept, but quite drastic.  Note that WriteCriticalBlock is called from vehicle code, so it opaque by the time it gets to us.

Possibly we could have a "DataFlash.current_timestamp()" method and some way of manipulating the result of that if required.  Construction of the packets would move from AP_HAL::micros() to "DataFlash.current_timestamp()".

OXINARF, 2016-10-05:
@peterbarker I misunderstood what you said in the previous post. So, the problem is that the EKF sends MAVLink messages and those generate a dataflash message which will have a different time. I think an easy fix would be to add a method "DataFlash.freezeTime(time)" (and an unfreezeTime) and then use that time for the Log_Write_Message if different than 0. This is obviously a workaround, but I think it is better than making big changes.