ArduPilot/ardupilot #4821: Plane: Autotune for TECS

magicrub, 2016-09-12:
Autotune for TECS, perhaps a new autotune mode called TECSTUNE?
Discuss

magicrub, 2016-09-13:
Primary params that need to be tuned:
TECS_CLMB_MAX
TECS_SINK_MIN
TECS_SINK_MAX

if we can get those tuned then we're drastically improving the flight envelope. Everything else is fine tuning.

jinchengde, 2016-09-20:
do you have any theory for this autotune?

magicrub, 2016-09-20:
not yet, I just made this issue so we wouldn't forget to do it someday. It's definitely in demand.

palm369, 2016-10-10:
This should be all calculated automatically under certain conditions, then printed in MP as msg.
Results = (alt2 - alt1) / ( T2- T1)

State1 (alt1, T1) = start of fulfilling conditions
State2 (alt2, T2) = fulfilling for given time or altitude

Subroutine could be enabled via a new TECS_TUNE_MSG parameter, so one can disable it to dont spam the msg tab in MP.

This could be the conditions to fulfill:

TECS_CLMB_MAX
- climbing in FBWA for 10s using THR_MAX while pulling full elevator maintaining FBW_MIN (or Cruise)
(Correction for batt voltage to compensate for different charging states?)

TECS_SINK_MIN
- Sinking in FBWA with no throttle while maintaining FBW_MIN for 10s
(whats about thermals?)

TECS_SINK_MAX
-  flying in FBWA with no throttle and pushing full elevator for 100m altitude
(Not every airframe can overspeed, eg foamies usually cant do it due to low weight/drag ratio)

EShamaev, 2022-06-06:
Tom, are you currently working on this feature?
I plan to implement it, but if you are on the way then no need to do double job.

magicrub, 2025-09-06:
Ryan is

Ryanf55, 2025-09-08:
> Max climb rate at given airspeed can be estimated in level flight based on horizontal acceleration (based on power excess causing acceleration).
> 
> `V_VMax=V_A*a_X/g+V_V`
> 
> `V_A` is airspeed `a_X` is forward acceleration `g` is gravity acceleration `V_V` is vertical velocity `V_VMax` is max climb rate at current airspeed.

Nice, this could be a good way to seed the initial guess too. This would be done at THR_MAX, then measure the peak acceleration until the vehicle hits AIRSPEED_MAX, then allow it to rotate up into the calculated climb rate and refine?

Georacer, 2026-04-07:
Thanks for sharing @RYOYA0408 ! Even though it's not suitable for mass release, your algorithm contains some very good ideas!