ArduPilot/ardupilot #4825: Vibration failsafe actions

lvale, 2016-09-13:
@rmackay9 and I were discussing a recent event of a user vehicle that misbehaved due to very high vibration levels.

The initial idea would be to detect that vibration problem (constant/very high clipping) and assume a "failsafe" like action. This would imply several actions, like switching to a baro controlled height, reducing gains, beeping an alarm pattern, message to the GCS, and perhaps going on a althold/poshold height (at RTL height perhaps)

auturgy, 2019-01-27:
No reason to close. It’s a valid enhancement request.

rmackay9, 2019-01-28:
Yes, I definitely think we should keep this one..

rmackay9, 2019-02-28:
I think the new vibration failsafe could detect a problem using some or all of these criteria:

- EKF's climb rate and change in altitude don't agree.  Compare EKF's change in altitude vs climb rate over the same period.  If altitude is climbing but climb rate says we are falling then the EKF is confused
- check for change in primary IMU's clipping number (be careful of the main IMU changing)
- check if vibration levels are over some minimum (like 30m/s/s)
- all the above persists for X seconds (perhaps 2 seconds)

I think we can help the EKF deal with high vibration by changing the EK2_ALT_M_NSE parameter as described on [user EKF wiki page](http://ardupilot.org/copter/docs/common-apm-navigation-extended-kalman-filter-overview.html).  We would likely need to add a method to the AHRS class (which would feed down into the EKF2 and EKF3) which would take a scaler (perhaps from 0 to 1) which specifies how much to weight the accelerometers vs baro.  Then apply some kind of linear scaling between a min and max EK2_ALT_M_NSE value and then push this value into the EKFs.

We could test this new failsafe works by placing a buzzer directly a flight controller that does not include internal vibration isolation (like a Pixhawk1) and then trigger the buzzer with a new auxiliary function or perhaps by using the lost-copter alarm

rmackay9, 2020-01-27:
Copter-4.0 includes a vibration failsafe so I think I will close this issue.  If we need to re-open we can.

closed 2020-01-27 by rmackay9