ArduPilot/ardupilot #4826: Copter randomly oscillates in flight, Throttle would not lower when landed.

benjammin1988, 2016-09-13:
Using 3.4rc beta copter, my copter was experiencing random oscillations in flight.  I went to land the copter, and when I landed the motors would not spool down and the copter eventually tipped over and caused a lot of damage.  Noted that I was getting bad gyro health erros as well as compass  variance errors  at the end.

AirFrame: Stark Hx8 AG Sprayer Drone

Hardware: Pixhawk

DropBox Link to logs:

rmackay9, 2016-09-17:
Thanks for beta testing AC3.4.

There's quite a large and persistent motor imbalance on the vehicle.  In particular motor 3 (front left) is working very hard while motor 6 (back left) is much less stressed.  It's fairly common to see motor imbalances of 100pwm on a vehicle but on this vehicle it's about 300 which is about 35% of the entire range so it's quite big.  Maybe it would be good to double check the propeller directions and center-of-gravity of the vehicle.

I'm not exactly sure if this is related to the troubles the vehicle had detecting the landing.  It's possibly not related.  Basically it looks like it's target position was off from where it actually touched down so even while on the ground it was trying to move horizontally.  I guess the pilot wasn't trying to move the vehicle just before it touched down?  It could also be related to another "drift loiter" that's being reported by some other beta testers.  http://discuss.ardupilot.org/t/loiter-drift-in-3-4rc3

By the way, for the beta testing, it would be best to put the logs into the Copter-3.4 area in the forums instead of here in gitter.  http://discuss.ardupilot.org/c/arducopter/copter34

In general we save gitter for enhancement requests and bugs but only after we've done the analysis to determine if it's really a bug or something else.

I'm going to close this issue if that's ok but I'll still see replies that appear here.

closed 2016-09-17 by rmackay9

benjammin1988, 2016-09-18:
Thank you Randy..  The CoG was off that day and the problem was fixed..
When I uploaded this log to arduplotter, I saw IMU state reset messages,
which after digging around, may have been a result of a faulty voltage
regulator to my IMU.  I switched out the pixhawk and the random osillations
in flight disappeared as well as the IMU state reset messages. Walah!  It
was pretty scary having it land and then try to take back off as this is
about a 150lb x8 Herbicide spraying drone

I just sent ardupilot initiative $100.00..  I plan on donating more in next
couple weeks to reach the $1,000.00 required to be a partner on your
website and go to the annual meeting mentioned.. I hope this is okay!

best,

Ben

magicrub, 2016-09-18:
>> just sent ardupilot initiative $100.00..  I plan on donating more in next couple weeks to reach the $1,000.00 required to be a partner

Did you mean Ardupilot.org? The initiative is actually a separate for-profit consulting service.

benjammin1988, 2016-09-19:
Got it well I sent it to initiative, whoops!

Ben

rmackay9, 2016-09-19:
@benjammin1988, ah, great that the problems was solved.

Re the donation, AI can transfer the $100 to ardupilot.org.  The only issue might be the non-profit reporting for you of this donation 'cuz ardupilot.org is a non-profit while AI is not.

benjammin1988, 2016-09-26:
Hey Randy,

Just wondering how did you check the pwm output for the motors when you
reviewed my last log?  This type of review would be useful for us as we are
building big herbicide spraying drones and COG will always be very
important.

Thanks for your time!

Ben

magicrub, 2016-09-26:
> t/loiter-drift-in-3-4rc3

RCOUT in the dataflash bin file on the sd card.