ArduPilot/ardupilot #4827: EKF Failsafe in Land mode and FS_EKF_ACTION set != 3 doesn't stop using position controller

John-Ingersoll, 2016-09-13:
#### Issue details
While flying a ArduCopter 3.4rc1 one-off, we encountered a battery failsafe and the vehicle started to land.  During land, we encountered a second failsafe for the EKF.  Even though the EKF was bad, the position controller didn't stop running.  This lead to the bad EKF position driving the vehicle into a fly away condition and it crashed.

ekfcheck.cpp: void Copter::failsafe_ekf_event(): If a EKF failsafe occurs in land, the do nothing check in the beginning won't kick over to a land without GPS.
```
if (!mode_requires_GPS(control_mode) && (g.fs_ekf_action != FS_EKF_ACTION_LAND_EVEN_STABILIZE)) {
        return;
 }
```
The fs_ekf_action I had selected was 1: Land so this check returned with no action taken.
This needs to allow a switchover to land_do_not_use_GPS();

In addition, this interferes with the failsafe.ekf flag being toggled with the execution of ekf_check().

#### Version
ArduCopter 3.4rc

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Home built Hexa copter frame

#### Hardware type
PixHawk v2

rmackay9, 2016-09-14:
John,
Thanks very much for this report and detailed analysis!  I'll check it out for sure.

rmackay9, 2016-10-10:
The basic issue is resolved with the above commit.

I have a follow-up question though (also posted on Gitter) re how the EKF failsafe should act if it triggers while the vehicle is in LAND mode but the user has FS_EKF_ACTION set to "2:AltHold".  As it is now, it will switch to AltHold (as the action specifies) but I think it probably makes sense that it instead switch to pilot controlled LAND (i.e. non GPS LAND).  Does everyone agree?

Thanks again to John-Ingersoll for the detailed report!

closed 2016-10-10 by rmackay9