ArduPilot/ardupilot #4828: Copter: Pilot should also be able to manually disarm in case of emergency

John-Ingersoll, 2016-09-13:
#### Issue details
During a set of recent crashes on a ArduCopter 3.4rc1 one-off build, the pilot was unable to disarm the hexa-copter from the RC.  The motors failed to spin down upon touchdown from a failsafe forced landing.  The motors spun up and the helicopter skipped across the room while in LAND mode.  Luckily no one was in the flight path so no harm was done, but it could be extremely dangerous.  The pilot expected to be able to hold the throttle down and disarm the vehicle but ArduCopter was unresponsive.

There is the "FS_THR_ENABLE" parameter which allows a failsafe to be triggered from the RC.  And the loss of communications from the RC also triggers a failsafe.  These only trigger a failsafe event such as land or alt hold.

The pilot should always have a quick option to disarm the helicopter in case of an emergency (one action).

Because the failsafe action was to switch to LAND mode, holding the throttle down in the land mode did not disarm the motors.

events.cpp: bool Copter::should_disarm_on_failsafe()
`bool Copter::should_disarm_on_failsafe() {
    switch(control_mode) {
        case STABILIZE:
        case ACRO:
            // if throttle is zero OR vehicle is landed disarm motors
            return ap.throttle_zero || ap.land_complete;
            break;
        case AUTO:
            // if mission has not started AND vehicle is landed, disarm motors
            return !ap.auto_armed && ap.land_complete;
            break;
        default:
            // used for AltHold, Guided, Loiter, RTL, Circle, Drift, Sport, Flip, Autotune, PosHold
            // if landed disarm
            return ap.land_complete;
            break;
    }
}`

This should also list LAND.

#### Version
ArduCopter 3.4rc1

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Hand made Hexa-Copter

#### Hardware type
PixHawk v2

lvale, 2016-09-13:
For the case of an emergency there is a EMERGENCY STOP function that cuts of the motors immediately.

rmackay9, 2016-09-14:
I'm sure it's all a bit scary when the vehicle can't be quickly disarmed.  The underlying issue, as you've hinted, is it didn't detect the landing quickly.  If it doesn't think it's landed it won't allow disarming the vehicle just from the sticks (because it's possible that a user could unintentionally perform the same stick movement while the vehicle is flying).

Here's the wiki page to the auxiliary switch setup: http://ardupilot.org/copter/docs/channel-7-and-8-options.html.  The "Emergency Stop Motors" is listed there along with a little video.

We have other options as well so the GCS can also be used to kill the motors (see MP's Flight Data > Actions tab and it's Arm/Disarm button).

Switching to stabilize and pulling the throttle to zero will allow disarming.

I'm going to close this because we have a number of ways to do it.  Hope that's ok.

closed 2016-09-14 by rmackay9

John-Ingersoll, 2016-09-17:
Thanks.  I looked into this and it is there.  I informed our pilots and
they were happy to find this information.  Sorry for the distraction.