ArduPilot/ardupilot #4830: Copter Bug:

popwebz, 2016-09-14:
#### Issue details
Mavlink functionality is broken for Mavlink-enabled Alexmos Gimbal.
I have the gimbal connected to the Serial4
SERIAL4_BAUD = 115
SERIAL4_PROTOCOL = 1
MNT_TYPE = 4

I've experimented with other protocols with no success.

The gimbal is set to send a heartbeat to Pixhawk
Unfortunately the Alexmos board is reporting
AHRS-no, GPS-no, RC-no, Control–no
However it does receive empty packets from the Pixhawk as can be seen in the logs

#### Version
AC 3.3

#### Platform
[x] Copter

#### Hardware type
Pixhawk

#### Logs from the gimbal:
ML1 out: 66 [9f 1 9a ] [5 0 1 1 6 1 ]
ML1 out: 0 [a0 1 9a ] [0 0 0 0 1a 0 10 4 3 ]
ML1 in: 0 [8f 1 1 ] [0 0 0 0 2 3 51 3 3 ]
ML1 out: 0 [a1 1 9a ] [0 0 0 0 1a 0 10 4 3 ]
ML1 in: 0 [90 1 1 ] [0 0 0 0 2 3 51 3 3 ]
ML1 out: 66 [a2 1 9a ] [14 0 1 1 a 1 ]

This continues on. As can be seen, the Pixhawk is only sending zeros to the gimbal.

OXINARF, 2016-12-20:
This isn't a bug, Alexmos support in ArduPilot is limited to the serial protocol. Feel free to contribute the MAVLink protocol though.

peterbarker, 2024-07-31:
I don't  think there's demand for this.

I also can't see what an "Alexmos mavlink protocol is"...

@brandonkal were you still interested in this?

brandonkal, 2024-07-31:
No that was 8 years ago. Closing

closed 2024-07-31 by brandonkal