ArduPilot/ardupilot #4849: read_rangefinder task can take up to 1.4ms with LightwareI2C LIDAR

jschall, 2016-09-16:
#### Issue details
I am investigating timing issues in ArduCopter, and the read_rangefinder task is taking up to 1.4ms when a Lightware SF11/C is connected via I2C.

#### Version
19d580d60fee5b95e0b13f02e66adc823da29c4d

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quad

#### Hardware type
3DR Pixhawk

#### Logs
n/a

jschall, 2016-09-16:
Other I2C devices on the bus:
PixyCam
HMC5983

The way the AP_HAL::I2CDevice interface is done, it seems like transactions have to be blocking? That's not good...

rmackay9, 2016-09-17:
I think perhaps this is a duplicate of this issue: https://github.com/ArduPilot/ardupilot/issues/4803

closed 2016-09-17 by rmackay9