ArduPilot/ardupilot #4853: Copter: EKF should report position shift if EKF core changes

rmackay9, 2016-09-19:
In Copter-3.4 we have multiple EKF cores.  It's possible the active core may change and the position estimate could move by several meters.  The EKF should report to the upper controllers that the position has shifted (and the amount of the shift) so that the position controllers can decide whether to re-adjust the target or not.

OXINARF, 2016-09-20:
I know the plan for dual GPS is to have multiple instances of the EKF2, but I'm not seeing us doing that soon (since Pixhawk has its limits). So, with this report in place, maybe we should make the EKF2 switch between GPS (it already knows how to see if a GPS is failing/glitching) and it could then take that into account for itself and report back to the controller.

rmackay9, 2017-02-07:
This is included in AC3.4.4

closed 2017-02-07 by rmackay9