ArduPilot/ardupilot #4855: compile error in AP_GPS_NMEA.cpp in SITL

jinchengde, 2016-09-20:
#### Issue details
Waf: Leaving directory `/home/kent/Git/ardupilot/build/sitl'
Build failed
-> task in 'objs/AP_GPS' failed (exit status 1):
{task 49664400: cxx AP_GPS_NMEA.cpp -> AP_GPS_NMEA.cpp.0.o}
['/usr/lib/ccache/g++', '-std=gnu++11', '-fdata-sections', '-ffunction-sections', '-fno-exceptions', '-fsigned-char', '-Wall', '-Wextra', '-Wformat', '-Wshadow', '-Wpointer-arith', '-Wcast-align', '-Wundef', '-Wno-unused-parameter', '-Wno-missing-field-initializers', '-Wno-reorder', '-Wno-redundant-decls', '-Wno-unknown-pragmas', '-Werror=format-security', '-Werror=array-bounds', '-Werror=uninitialized', '-Werror=init-self', '-Wfatal-errors', '-Werror=unused-but-set-variable', '-O3', '-include', 'ap_config.h', '-Ilibraries', '-Ilibraries/GCS_MAVLink', '-I.', '-I../../libraries', '-I../../libraries/AP_Common/missing', '-DSKETCHBOOK="/home/kent/Git/ardupilot"', '-DCONFIG_HAL_BOARD=HAL_BOARD_SITL', '-DCONFIG_HAL_BOARD_SUBTYPE=HAL_BOARD_SUBTYPE_NONE', '../../libraries/AP_GPS/AP_GPS_NMEA.cpp', '-c', '-o/home/kent/Git/ardupilot/build/sitl/libraries/AP_GPS/AP_GPS_NMEA.cpp.0.o']

the detail information is:
../../libraries/AP_GPS/AP_GPS_NMEA.cpp: In static member function ‘static int32_t AP_GPS_NMEA::parsedecimal_100(const char*)’:
../../libraries/AP_GPS/AP_GPS_NMEA.cpp:194:22: error: ‘INT32_MAX’ was not declared in this scope
compilation terminated due to -Wfatal-errors.

#### Version
SHA-1: 035937ea7eb454fa455dbf30e5a3f18f324cc4c6

* AP_GPS: add some safety checks to _parse_decimal_100()

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover

#### Airframe type
flying wing

#### Hardware type
pixhawk

#### Logs
none

jinchengde, 2016-09-23:
It looks like that it happened just in Debian 7 system. when we upgrade to Debian 8, everything is OK
I guess maybe some relationship with gcc?

OXINARF, 2016-12-13:
We compile SITL without any issue. Also @jinchengde doesn't have a problem anymore, so I'm closing this.

closed 2016-12-13 by OXINARF

jinchengde, 2016-12-14:
Thanks! no issue now

wangxuanjie, 2017-03-30:
Hello @jinchengde , i have the similar problem too, and how you solve the problem